kismetwireless/kismet #439: failed to capture droneID and telemetry data using kismet

astro12345, 2022-10-11:
Hi,
i'm currently trying to capture the telemetry data ( latitude, longitude, altitude, velocity... etc) which are included in the wifi DroneID frame of some drones (in my case the Mavic air, the Mavic pro 2 and the phantom). Using kismet, i can only detect the drones as WIFI access points with UAV fingerprint (with a little help from the kismet_uav.conf file) but never as droneI. In addition to that, all the telemetry data is null for some reason. I would like to know if there is any specific configuration i must use in order to detect the wifi DroneID. 
here is a screenshot of what i get in the kimset localhost interface:
[image]

thank you in advance for your precious help.

yiquchangge1, 2023-10-09:
Hello,I also encountered the same problem as you. I use Mavci air UAV and Kismet for capture. MAVIC AIR wifi can be obtained, but it does not classify UAV/Drone。
[image]

Black-1ron, 2024-10-05:
[astro12345]or [yiquchangge1]. Do you have new information about that? i am trying the same capture the telemetry and i get the same.
In Debug option i can some information but not about telemetry 

[image]