kismetwireless/kismet #448: Configure/Make errors on Mac OS

elimbaum, 2022-10-28:
I am trying to install Kismet from source on mac OS, following the [quick start instructions](https://www.kismetwireless.net/docs/readme/quickstart/).

I ran into a few hiccups linking libraries in; I'm not sure if this is something specific to my machine, or something to do with homebrew.

<details>
<summary>System Information</summary>

I'm on macOS Monterey, 12.6.1.

```
$ brew --version
Homebrew 3.6.7
Homebrew/homebrew-core (git revision 1507d2853b2; last commit 2022-10-28)
Homebrew/homebrew-cask (git revision 23ab0f2d6b; last commit 2022-10-28)
$ brew --env
HOMEBREW_CC: clang
HOMEBREW_CXX: clang++
MAKEFLAGS: -j12
CMAKE_PREFIX_PATH: /usr/local
CMAKE_INCLUDE_PATH: /Library/Developer/CommandLineTools/SDKs/MacOSX12.sdk/System/Library/Frameworks/OpenGL.framework/Versions/Current/Headers
CMAKE_LIBRARY_PATH: /Library/Developer/CommandLineTools/SDKs/MacOSX12.sdk/System/Library/Frameworks/OpenGL.framework/Versions/Current/Libraries
PKG_CONFIG_LIBDIR: /usr/lib/pkgconfig:/usr/local/Homebrew/Library/Homebrew/os/mac/pkgconfig/12
HOMEBREW_GIT: git
HOMEBREW_SDKROOT: /Library/Developer/CommandLineTools/SDKs/MacOSX12.sdk
ACLOCAL_PATH: /usr/local/share/aclocal
PATH: /usr/local/Homebrew/Library/Homebrew/shims/mac/super:/usr/bin:/bin:/usr/sbin:/sbin
$ brew --prefix
/usr/local
$ uname -a
Darwin MM273306-PC 21.6.0 Darwin Kernel Version 21.6.0: Thu Sep 29 20:12:57 PDT 2022; root:xnu-8020.240.7~1/RELEASE_X86_64 x86_64
```
</details>

Originally, I couldn't run `./configure`. The quickstart suggests

```
$ LDFLAGS=-L$(brew --prefix)/lib CPPFLAGS="-I$(brew --prefix)/include -I$(brew --prefix openssl)/include" ./configure --with-openssl=$(brew --prefix openssl)
```

but this gave me some errors:
```
configure: WARNING: unrecognized options: --with-openssl
cat: VERSION: No such file or directory
... other messages ...
configure: error: Package requirements (libwebsockets >= 3.1.0) were not met:

No package 'libwebsockets' found
```

I'm assuming the first two lines aren't a problem, but we definitely need `libwebsockets`. The same happened for `protobuf`.

Adding the `PKG_CONFIG_PATH` environment variable to `./configure` fixed it:

```
PKG_CONFIG_PATH="$(brew --prefix libwebsockets)/lib/pkgconfig:$(brew --prefix protobuf)/lib/pkgconfig" \
LDFLAGS=-L$(brew --prefix)/lib \
CPPFLAGS="-I$(brew --prefix)/include -I$(brew --prefix openssl)/include" \
./configure --with-openssl=$(brew --prefix openssl)
```

> It took me a while to realize that colon `:` is used to separate paths, not spaces.

I'm not sure why this is necessary, because both `libwebsockets` and `protobuf` are installed in `/usr/local/opt/`, as expected.

After taking this step, `make` and installation worked fine – Kismet is now running perfectly.

96611229, 2023-01-04:
This fails for me when installing via brew, looks like same issue like yours though not sure how to fix it.

```
==> Installing kismetwireless/kismet/kismet
==> ./configure --prefix=/usr/local/Cellar/kismet/2022-08 --libdir=/usr/local/Cellar/kismet/2022-08/lib --enable-wifi-co
==> make install
Last 15 lines from /Users/user/Library/Logs/Homebrew/kismet/02.make:
protoc-c --c_out=./protobuf_c/ -I protobuf_definitions/ protobuf_definitions/linuxbluetooth.proto
protoc-c --c_out=./protobuf_c/ -I protobuf_definitions/ protobuf_definitions/eventbus.proto
(cd capture_bladerf_wiphy && /Library/Developer/CommandLineTools/usr/bin/make)
clang -MM -Wall -Wno-unknown-warning-option -Wno-format-truncation -Wno-unused-function -g -I. -fPIE   -I/usr/local/opt/openssl@3/include -pthread -I/usr/local/Cellar/protobuf-c/1.4.1_1/include -I/usr/local/Cellar/libpcap/1.10.2/include  -I/usr/local/opt/openssl@3/include -I/usr/local/opt/openssl@3/include  -I/usr/local/opt/openssl@3/include  ../wifi_ht_channels.c | sed -e "s/\.o/\.c.o/" > ../wifi_ht_channels.c.d
clang -MM -Wall -Wno-unknown-warning-option -Wno-format-truncation -Wno-unused-function -g -I. -fPIE   -I/usr/local/opt/openssl@3/include -pthread -I/usr/local/Cellar/protobuf-c/1.4.1_1/include -I/usr/local/Cellar/libpcap/1.10.2/include  -I/usr/local/opt/openssl@3/include -I/usr/local/opt/openssl@3/include  -I/usr/local/opt/openssl@3/include  capture_bladerf_wiphy.c | sed -e "s/\.o/\.c.o/" > capture_bladerf_wiphy.c.d
make[2]: *** No rule to make target `../libkismetdatasource.a', needed by `kismet_cap_bladerf_wiphy'.  Stop.
make[2]: *** Waiting for unfinished jobs....
In file included from capture_bladerf_wiphy.c:69:
./../capture_framework.h:62:10: fatal error: 'protobuf_c/kismet.pb-c.h' file not found
#include "protobuf_c/kismet.pb-c.h"
         ^~~~~~~~~~~~~~~~~~~~~~~~~~
1 error generated.
make[1]: *** [capture_bladerf_wiphy/kismet_cap_bladerf_wiphy] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [install] Error 2

Do not report this issue to Homebrew/brew or Homebrew/core!

```

rob-scheepens, 2023-02-10:
"brew install kismet" failing for me with the same end result, but different output:
```
==> Installing kismet from kismetwireless/kismet
==> ./configure --prefix=/usr/local/Cellar/kismet/2022-08 --libdir=/usr/local/Cellar/kismet/2022-08/lib --enable-wifi-coconut --enable-bladerf --disable-python-tools --with-openssl=/usr/local/opt/openssl@3
==> make install
Last 15 lines from /Users/robscheepens/Library/Logs/Homebrew/kismet/02.make:
clang++ -std=gnu++17 -MM -MP -Wall -Wno-unknown-warning-option -Wno-deprecated-declarations -Wno-format-truncation -Wno-unused-local-typedefs -Wno-unused-function -Wno-infinite-recursion -g -I. -fPIE -g -O2 -O3 -I/usr/local/Cellar/protobuf/21.12/include -DKS_STR_ENCODING_NONE -I/usr/local/Cellar/libpcap/1.10.3/include -I/usr/local/opt/openssl@3/include -I/usr/local/opt/openssl@3/include  -MT protobuf_cpp/http.pb.cc.o protobuf_cpp/http.pb.cc -MF protobuf_cpp/http.pb.cc.d
clang++ -std=gnu++17 -MM -MP -Wall -Wno-unknown-warning-option -Wno-deprecated-declarations -Wno-format-truncation -Wno-unused-local-typedefs -Wno-unused-function -Wno-infinite-recursion -g -I. -fPIE -g -O2 -O3 -I/usr/local/Cellar/protobuf/21.12/include -DKS_STR_ENCODING_NONE -I/usr/local/Cellar/libpcap/1.10.3/include -I/usr/local/opt/openssl@3/include -I/usr/local/opt/openssl@3/include  -MT protobuf_cpp/datasource.pb.cc.o protobuf_cpp/datasource.pb.cc -MF protobuf_cpp/datasource.pb.cc.d
clang++ -std=gnu++17 -MM -MP -Wall -Wno-unknown-warning-option -Wno-deprecated-declarations -Wno-format-truncation -Wno-unused-local-typedefs -Wno-unused-function -Wno-infinite-recursion -g -I. -fPIE -g -O2 -O3 -I/usr/local/Cellar/protobuf/21.12/include -DKS_STR_ENCODING_NONE -I/usr/local/Cellar/libpcap/1.10.3/include -I/usr/local/opt/openssl@3/include -I/usr/local/opt/openssl@3/include  -MT protobuf_cpp/linuxbluetooth.pb.cc.o protobuf_cpp/linuxbluetooth.pb.cc -MF protobuf_cpp/linuxbluetooth.pb.cc.d
clang++ -std=gnu++17 -MM -MP -Wall -Wno-unknown-warning-option -Wno-deprecated-declarations -Wno-format-truncation -Wno-unused-local-typedefs -Wno-unused-function -Wno-infinite-recursion -g -I. -fPIE -g -O2 -O3 -I/usr/local/Cellar/protobuf/21.12/include -DKS_STR_ENCODING_NONE -I/usr/local/Cellar/libpcap/1.10.3/include -I/usr/local/opt/openssl@3/include -I/usr/local/opt/openssl@3/include  -MT protobuf_cpp/eventbus.pb.cc.o protobuf_cpp/eventbus.pb.cc -MF protobuf_cpp/eventbus.pb.cc.d
clang++ -std=gnu++17 -MM -MP -Wall -Wno-unknown-warning-option -Wno-deprecated-declarations -Wno-format-truncation -Wno-unused-local-typedefs -Wno-unused-function -Wno-infinite-recursion -g -I. -fPIE -g -O2 -O3 -I/usr/local/Cellar/protobuf/21.12/include -DKS_STR_ENCODING_NONE -I/usr/local/Cellar/libpcap/1.10.3/include -I/usr/local/opt/openssl@3/include -I/usr/local/opt/openssl@3/include  -MT util.cc.o util.cc -MF util.cc.d
clang++ -std=gnu++17 -MM -MP -Wall -Wno-unknown-warning-option -Wno-deprecated-declarations -Wno-format-truncation -Wno-unused-local-typedefs -Wno-unused-function -Wno-infinite-recursion -g -I. -fPIE -g -O2 -O3 -I/usr/local/Cellar/protobuf/21.12/include -DKS_STR_ENCODING_NONE -I/usr/local/Cellar/libpcap/1.10.3/include -I/usr/local/opt/openssl@3/include -I/usr/local/opt/openssl@3/include  -MT crc32.cc.o crc32.cc -MF crc32.cc.d
clang++ -std=gnu++17 -MM -MP -Wall -Wno-unknown-warning-option -Wno-deprecated-declarations -Wno-format-truncation -Wno-unused-local-typedefs -Wno-unused-function -Wno-infinite-recursion -g -I. -fPIE -g -O2 -O3 -I/usr/local/Cellar/protobuf/21.12/include -DKS_STR_ENCODING_NONE -I/usr/local/Cellar/libpcap/1.10.3/include -I/usr/local/opt/openssl@3/include -I/usr/local/opt/openssl@3/include  -MT macaddr.cc.o macaddr.cc -MF macaddr.cc.d
clang++ -std=gnu++17 -MM -MP -Wall -Wno-unknown-warning-option -Wno-deprecated-declarations -Wno-format-truncation -Wno-unused-local-typedefs -Wno-unused-function -Wno-infinite-recursion -g -I. -fPIE -g -O2 -O3 -I/usr/local/Cellar/protobuf/21.12/include -DKS_STR_ENCODING_NONE -I/usr/local/Cellar/libpcap/1.10.3/include -I/usr/local/opt/openssl@3/include -I/usr/local/opt/openssl@3/include  -MT uuid.cc.o uuid.cc -MF uuid.cc.d
clang++ -std=gnu++17 -MM -MP -Wall -Wno-unknown-warning-option -Wno-deprecated-declarations -Wno-format-truncation -Wno-unused-local-typedefs -Wno-unused-function -Wno-infinite-recursion -g -I. -fPIE -g -O2 -O3 -I/usr/local/Cellar/protobuf/21.12/include -DKS_STR_ENCODING_NONE -I/usr/local/Cellar/libpcap/1.10.3/include -I/usr/local/opt/openssl@3/include -I/usr/local/opt/openssl@3/include  -MT xxhash.cc.o xxhash.cc -MF xxhash.cc.d
clang++ -std=gnu++17 -MM -MP -Wall -Wno-unknown-warning-option -Wno-deprecated-declarations -Wno-format-truncation -Wno-unused-local-typedefs -Wno-unused-function -Wno-infinite-recursion -g -I. -fPIE -g -O2 -O3 -I/usr/local/Cellar/protobuf/21.12/include -DKS_STR_ENCODING_NONE -I/usr/local/Cellar/libpcap/1.10.3/include -I/usr/local/opt/openssl@3/include -I/usr/local/opt/openssl@3/include  -MT boost_like_hash.cc.o boost_like_hash.cc -MF boost_like_hash.cc.d
clang++ -std=gnu++17 -MM -MP -Wall -Wno-unknown-warning-option -Wno-deprecated-declarations -Wno-format-truncation -Wno-unused-local-typedefs -Wno-unused-function -Wno-infinite-recursion -g -I. -fPIE -g -O2 -O3 -I/usr/local/Cellar/protobuf/21.12/include -DKS_STR_ENCODING_NONE -I/usr/local/Cellar/libpcap/1.10.3/include -I/usr/local/opt/openssl@3/include -I/usr/local/opt/openssl@3/include  -MT sqlite3_cpp11.cc.o sqlite3_cpp11.cc -MF sqlite3_cpp11.cc.d
[... cut ...] 1.12/include -DKS_STR_ENCODING_NONE -I/usr/local/Cellar/libpcap/1.10.3/include -I/usr/local/opt/openssl@3/include -I/usr/local/opt/openssl@3/include  -MT globalregistry.cc.o globalregistry.cc -MF globalregistry.cc.d
make[1]: *** [capture_bladerf_wiphy/kismet_cap_bladerf_wiphy] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [install] Error 2

If reporting this issue please do so at (not Homebrew/brew or Homebrew/core):
  https://github.com/kismetwireless/homebrew-kismet/issues
```

kismetwireless, 2023-02-10:
Until the next release, you'll need to use the nightly recipes for brew, or compile the release manually.