kismetwireless/kismet #449: JSON Parse Error During Export

aeonik, 2022-11-01:
Hello,
I was trying to export a 10GB dump of a local capture using the following command:
`kismetdb_dump_devices --force --ekjson --in Kismet-20221028-19-33-03-1.kismet --out Kismet-20221028-19-33-03-1.json`

During the export the program crashes with the following error:
> ERROR:  Could not process device JSON: [json.exception.parse_error.101] parse error at line 1, column 1374: syntax error while parsing value - invalid string: ill-formed UTF-8 byte; last read: '"8l\u0014Le�P'ERROR:  Could not process device JSON: [json.exception.parse_error.101] parse error at line 1, column 1254: syntax error while parsing value - invalid string: ill-formed UTF-8 byte; last read: '"�'ERROR:  Could not process device JSON: [json.exception.parse_error.101] parse error at line 1, column 1257: syntax error while parsing value - invalid string: ill-formed UTF-8 byte; last read: '"�'ERROR:  Could not process device JSON: [json.exception.parse_error.101] parse error at line 1, column 1598: syntax error while parsing value - invalid string: ill-formed UTF-8 byte; last read: '"ɠ�'ERROR:  Could not process device JSON: [json.exception.parse_error.101] parse error at line 1, column 1374: syntax error while parsing value - invalid string: ill-formed UTF-8 byte; last read: '"�'ERROR:  Could not process device JSON: [json.exception.parse_error.101] parse error at line 1, column 1332: syntax error while parsing value - invalid string:

I was able to locate this record in the SQLite KismetDB, however the SQLite program that I was using also seemed to run into some decoding errors and only gave me a hexadecimal output.

Despite this, I believe I have located at least one of the problematic strings that was captured by Kismet that is causing the decoding errors:
`
"dot11.probedssid.wps_uuid_e": "8l\u0014LeÌPÒñ;Å\n±v_"`

The raw hex of the value is here (pulled from a hex dump of the file): `22 38 6C 5C 75 30 30 31 34 4C 65 CC 50 D2 85 F1 3B C5 5C 6E B1 76 5F 22`

The record seems to be completely intact, it just seems to be a decoding error.

Regards,
Aeonik

kismetwireless, 2022-11-01:
> ERROR: Could not process device JSON: [json.exception.parse_error.101] parse error at line 1, column 1374: syntax error while parsing value - invalid string: ill-formed UTF-8 byte; last read: '"8lLe�P'ERROR: Could not process device JSON: [json.exception.parse_error.101] parse error at line 1, column 1254: syntax error while parsing value - invalid string: ill-formed UTF-8 byte; last read: '"�'ERROR: Could not process device JSON: [json.exception.parse_error.101] parse error at line 1, column 1257: syntax error while parsing value - invalid string: ill-formed UTF-8 byte; last read: '"�'ERROR: Could not process device JSON: [json.exception.parse_error.101] parse error at line 1, column 1598: syntax error while parsing value - invalid string: ill-formed UTF-8 byte; last read: '"ɠ�'ERROR: Could not process device JSON: [json.exception.parse_error.101] parse error at line 1, column 1374: syntax error while parsing value - invalid string: ill-formed UTF-8 byte; last read: '"�'ERROR: Could not process device JSON: [json.exception.parse_error.101] parse error at line 1, column 1332: syntax error while parsing value - invalid string:
> "dot11.probedssid.wps_uuid_e": "8lLeÌPÒ�ñ;Å\n±v_"

Should be fixed in the git code as of a month or so ago and will be included in the next release.

Combination of changes to the encode and decode process. Basically comes down to how strict your decider is about utf8.

RFDeck, 2026-11-07:
`dot11.probedssid.wps_uuid_e` has gone through `munge_to_printable()` since 29bb81c1 (2022-08-14), which escapes
bytes that aren't valid UTF-8. That is in every release since 2023-07-R1, and current master doesn't store the field any
more. SSIDs and the WPS name fields are escaped the same way. Logs written by older versions can still contain the raw
bytes. For those, `kismetdb_dump_devices` prints an error and skips the device. Can this be closed?