kismetwireless/kismet #454: HTTP 400 error when using

peterlavelle, 2022-12-28:
Hi,

Trying to use the kismet_cap_sdr_rtl433 on a remote machine to connect to the Kismet master server I have running here (Using version 9999+git2022-12-28r754fe67d-1 on both the drone and master servers) but having issues

I'm using nginx on the Master server to terminate SSL and my own SSL cert. Nginx config looks something like this

server {
server_name kismet.batcave  ;
listen 443 ssl ;
ssl_certificate     /path/to/ssl.crt;
ssl_certificate_key /path/to/ssl.key;
location /kismet/ {
proxy_pass http://127.0.0.1:2501;
proxy_set_header Host $http_host;
proxy_set_header Upgrade $http_upgrade;
proxy_set_header Connection "upgrade";
proxy_set_header X-Real-IP $remote_addr;
proxy_set_header X-Forwarded-For $proxy_add_x_forwarded_for;
proxy_set_header X-Scheme $scheme;
proxy_set_header X-Proxy-Dir kismet/;
add_header X-Proxy-Dir kismet/;
proxy_http_version 1.1;
client_max_body_size 0;
}
        
}

And in kismet_site.conf, I have these options enabled

httpd_bind_address=127.0.0.1
httpd_uri_prefix=/kismet 
httpd_allow_cors = True 
httpd_allowed_origin=https://my.kismetmaster.hostname 
server_announce=false

Running the command on the Drone device below:

kismet_cap_sdr_rtl433 --connect kismetmasterhostname:443 --apikey=<redacted> --source=rtl433-0:name=mysdr --endpoint /kismet/datasource/remote/remotesource.ws --ssl

I get the error message:

Failed to connect to remote host:  server rejected WebSocket connection: HTTP 400

From what I can tell, it maybe trying to connect using HTTP, which Nginx rejects, despite having the --ssl option specified, but please correct me on that if I'm wrong

I did manage to capture a couple of errors in the master servers nginx logs (posted below with the API creds redacted)

2022/10/13 18:00:53 [debug] 2867628#2867628: *14 SSL_read: 138
2022/10/13 18:00:53 [debug] 2867628#2867628: *14 SSL_read: 136
2022/10/13 18:00:53 [debug] 2867628#2867628: *14 SSL_read: 118
2022/10/13 18:00:53 [debug] 2867628#2867628: *14 SSL_read: -1
2022/10/13 18:00:53 [debug] 2867628#2867628: *14 SSL_get_error: 2
2022/10/13 18:00:53 [debug] 2867628#2867628: *14 send: fd:17 392 of 392
2022/10/13 18:00:53 [debug] 2867628#2867628: *14 event timer: 17, old: 3461711451, new: 3461711455
2022/10/13 18:00:53 [debug] 2867628#2867628: *14 http run request: "/kismet/datasource/remote/remotesource.ws?KISMET=redacted"
2022/10/13 18:00:53 [debug] 2867628#2867628: *14 http upstream process upgraded, fu:1
2022/10/13 18:00:53 [debug] 2867628#2867628: *14 event timer: 17, old: 3461711451, new: 3461711455

kismetwireless, 2022-12-28:
Possibly addressed in some commits today; they'll go into the nightly builds tonight.

peterlavelle, 2023-01-07:
Ok, I've worked around this for now with the legacy remote capture port and a SSH tunnel. fwiw, I used [autossh](https://packages.ubuntu.com/search?keywords=autossh)  and a guide [here](https://handyman.dulare.com/ssh-tunneling-with-autossh/) to get the remote capture device to automatically bring up the Tunnel on boot and keep it up.