kismetwireless/kismet #456: API - Additional Documentation/Examples For Device Views Endpoint

goldensun441, 2023-01-12:
If possible it would be nice to get additional examples for accessing the device views endpoint, or any assistance using it would be appreciated. 

Reference Doc-
https://www.kismet-wifi.net/docs/api/device_views/#devices-by-view-and-time

_______________________________________________

What I'm Trying to do-
I'm trying to periodically poll the device views endpoint to gather new device sightings in chronologically-ordered chunks, so that I don't miss data and dont overload or bog down my request with oversized bulk data requests.

i.e.
'of all devices you've seen in ascending order of last_time,  give me the first 50 since my timestamp
then I update timestamp based on the latest last_time in the results, and can come back later with a new request without missing data.

*Example bare bones python of a request attached*
I've kept it simple in the file, but I've tried at least 50 different combinations of payloads/params already, and cannot get it to do what I want. Figured it was time to at least ask
_______________________________________________

What I'm Actually Getting-
field simplification within the json payload is the only thing i know for sure is working, it seems all post variables are ignored. it spits out a very long unordered list of all devices since the timestamp.

I've kept it simple in the file, but I've tried at least 50 different combinations of payloads/params over many hours already, and cannot get it to do what I expect.
_______________________________________________

Additional Notes-

I initially started with the python-kismet-rest library to read devices since a timestamp, but it does not support device views or pagination, despite the documentation encouraging all clients to use the device view api going forwards.

Not sure if I found the right section or not; but digging through the source a bit, I found some references within devicetracker_view.cc to some familiar parameters: ("colmap", "start", "length",  "search[value]"). It seems to only bother looking for these http_variables if the json().value("datatable", false), so I test with that value as false in my json payload.

[test_scripts.zip](https://github.com/kismetwireless/kismet/files/10397625/test_scripts.zip)

goldensun441, 2023-01-13:
Any working example of this endpoint's pagination in action from the point of a remote client would be good at this point- 
python, curl, raw html, ajax (maybe).

I would expect something as simple dropping this into my browser bar-
"http://<kismetuser>:<kismetpassword>@<IP>:2501/devices/views/all/last-time/0/devices.json?start=0&length=5&draw=1"
would at least limit the data entries returned to only the first 5, but it doesn't.

RFDeck, 2026-11-07:
`/devices/views/<view>/last-time/<ts>/devices.json` only filters by time. It ignores `start`, `length` and sort
parameters and always returns the whole list. Paging and sorting are on `/devices/views/<view>/devices.json`. Put
`page`, `size` (or `length`), `sort` and `sort_dir` in the URL. Put `fields` and the time filter `last_time` in the
JSON body. For example:

```
curl -u user:pass \
  --data-urlencode 'json={"fields": ["kismet.device.base.macaddr", "kismet.device.base.last_time"], "last_time": -600}' \
  'http://localhost:2501/devices/views/all/devices.json?page=0&size=5&sort=kismet.device.base.last_time&sort_dir=asc'
```

That returns `{"data": [...], "last_page": ..., "last_row": ..., "total_row": ...}`. `page` counts from 0. With
`sort_dir=asc`, the most recent devices come first, since the direction is named the other way round. Without
`sort_dir`, or with `desc`, the oldest come first. I checked this on 2026-10-R1. A negative `last_time` in the body only
works in master (#719); on 2026-10-R1, use an absolute timestamp. The `start`/`draw` parameters belong to the old
DataTables mode, which needs `"datatable": true` in the JSON body.