kismetwireless/kismet #457: No RTL_433 data in Kismet

Dinosaur-Ceiling, 2023-01-23:
Hello All,

Hoping to get some assistance with this issue. Been having problems for a while and cannot figure out how to fix it. This is a very similar problem to #222 (minus the use of docker) and my previous post #417 .

Any help would be appreciated!

I'm trying to run kismet with RTL_433 to visualize 315Mhz information (TMPS data). 

When I run rtl_433 using the following command on a laptop: 
rtl_433 -f 315M -F json:FILENAME 

I get a ton of data collected and outputted as follows: 

"time" : "2023-01-23 13:03:59", "model" : "Abarth 124 Spider", "type" : "TPMS", "id" : "c1c05b66", "flags" : "60", "pressure_kPa" : 207.000, "temperature_C" : 4.000, "status" : 43, "mic" : "CHECKSUM"}
{"time" : "2023-01-23 13:05:38", "model" : "Schrader-EG53MA4", "type" : "TPMS", "flags" : "4d930079", "id" : "94F0DB", "pressure_kPa" : 232.500, "temperature_F" : 81.000, "mic" : "CHECKSUM"}
{"time" : "2023-01-23 13:07:14", "model" : "Schrader-EG53MA4", "type" : "TPMS", "flags" : "4d930060", "id" : "647AFB", "pressure_kPa" : 377.500, "temperature_F" : 75.000, "mic" : "CHECKSUM"}
{"time" : "2023-01-23 13:07:14", "model" : "Schrader-EG53MA4", "type" : "TPMS", "flags" : "4d930060", "id" : "647AFB", "pressure_kPa" : 377.500, "temperature_F" : 75.000, "mic" : "CHECKSUM"}
{"time" : "2023-01-23 13:07:23", "model" : "Schrader-EG53MA4", "type" : "TPMS", "flags" : "4d930060", "id" : "692630", "pressure_kPa" : 517.500, "temperature_F" : 76.000, "mic" : "CHECKSUM"}
{"time" : "2023-01-23 13:07:24", "model" : "Schrader-EG53MA4", "type" : "TPMS", "flags" : "4d930060", "id" : "692630", "pressure_kPa" : 517.500, "temperature_F" : 76.000, "mic" : "CHECKSUM"}
{"time" : "2023-01-23 13:08:00", "model" : "Toyota", "type" : "TPMS", "id" : "f1a5345b", "status" : 128, "pressure_PSI" : 33.500, "temperature_C" : 5.000, "mic" : "CRC"}
{"time" : "2023-01-23 13:08:09", "model" : "Toyota", "type" : "TPMS", "id" : "f1a52ee8", "status" : 128, "pressure_PSI" : 32.750, "temperature_C" : 4.000, "mic" : "CRC"}
{"time" : "2023-01-23 13:10:10", "model" : "Toyota", "type" : "TPMS", "id" : "d286d1bc", "status" : 128, "pressure_PSI" : 32.750, "temperature_C" : 15.000, "mic" : "CRC"}
{"time" : "2023-01-23 13:12:59", "model" : "Toyota", "type" : "TPMS", "id" : "d997ce1c", "status" : 130, "pressure_PSI" : 34.500, "temperature_C" : 7.000, "mic" : "CRC"}
{"time" : "2023-01-23 13:12:59", "model" : "Toyota", "type" : "TPMS", "id" : "d997ce1c", "status" : 131, "pressure_PSI" : 34.500, "temperature_C" : 7.000, "mic" : "CRC"}
{"time" : "2023-01-23 13:13:08", "model" : "Toyota", "type" : "TPMS", "id" : "d99805ce", "status" : 128, "pressure_PSI" : 33.500, "temperature_C" : 6.000, "mic" : "CRC"}
{"time" : "2023-01-23 13:15:04", "model" : "Toyota", "type" : "TPMS", "id" : "d9ad3312", "status" : 128, "pressure_PSI" : 26.750, "temperature_C" : 5.000, "mic" : "CRC"}
{"time" : "2023-01-23 13:15:08", "model" : "Toyota", "type" : "TPMS", "id" : "d9ad331c", "status" : 128, "pressure_PSI" : 26.750, "temperature_C" : 7.000, "mic" : "CRC"}
{"time" : "2023-01-23 13:18:01", "model" : "Abarth 124 Spider", "type" : "TPMS", "id" : "c065063d", "flags" : "60", "pressure_kPa" : 162.840, "temperature_C" : 3.000, "status" : 26, "mic" : "CHECKSUM"}

But if I try with kismet, I never get anything. I've tried with multiple SDR's to make sure. Also tried running both at the same time on two devices. The one running RTL_433 by itself picks up data but the one running Kismet gets nothing. Seems like RTL_433 works via command line but not in kismet.

My kismet_site.conf contains the following:

source=rtl433-0:channel=315MHz,name=TPMS

Am I doing something wrong? Not sure why Kismet is not seeing and populating this data from RTL_433.

Dinosaur-Ceiling, 2023-01-23:
From further testing, it seems like Kismet is defaulting the rtl_433 to run at 433 no matter what channel is set in the kimset.conf or kismet_site.conf files.

This was determined by running Kismet with SDR interface on and set with the following in the kismet.conf: 
source=rtl433-0:type=rtl433,channel=315MHz,name=TPMS

Ran rtl_433 -f 315M on a separate device. 
This device picks up all the TPMS as indicated above, but some of these devices appear to be broadcasting at both 315Mhz and 433MHz which was displayed on this device and the Kismet Interface. 

Additionally, the Kismet interface listed the channel as 433MHz in the Kismet GUI. 

Not sure why it appears to be overriding the channel set in the conf files. Any ideas on how to correct this?

detectingdragon77, 2023-03-22:
Yeah having same issue, tried specifying the freq in /etc/kismet/kismet.conf, didn't work.

gfraysse, 2024-09-21:
I have a similar issue. One workaround is to create a "fake" rtl_433 script with the right command line arguments. Example on my Raspbian:
### rename the binary 
```
mv /usr/local/bin/rtl_433 /usr/local/bin/rtl_433.orig
```

create a new shell script "/usr/local/bin/rtl_433" with:
```bash
#!/bin/bash

/usr/local/bin/rtl_433.orig # <- your command line parameters here
```

### give the execution rights
```
sudo chmod a+x /usr/local/bin/rtl_433
```
### restart kismet

it will run the new rtl_433  bash script

et voilà :)