kismetwireless/kismet #458: Error when Installing via Brew

aero-oli, 2023-01-26:
get the following error when installing via brew on M1 Mac Pro with Ventura.

`brew install kismet
Running `brew update --auto-update`...
==> Auto-updated Homebrew!
Updated 2 taps (homebrew/core and homebrew/cask).
==> New Formulae
clang-build-analyzer
==> New Casks
rive

You have 1 outdated cask installed.
You can upgrade it with brew upgrade
or list it with brew outdated.

==> Fetching kismetwireless/kismet/kismet
==> Downloading https://github.com/kismetwireless/kismet/archive/refs/tags/kismet-2022-08-R1.zip
Already downloaded: /Users/oliververity/Library/Caches/Homebrew/downloads/08b3a742618b7cf9883ae51d1440ff5a2c042a7ab0bfbd6d6aa84d90595bd4ba--kismet-kismet-2022-08-R1.zip
==> Installing kismet from kismetwireless/kismet
==> ./configure --prefix=/opt/homebrew/Cellar/kismet/2022-08 --libdir=/opt/homebrew/Cellar/kismet/2022-08/lib --enable-wifi
==> make install
Last 15 lines from /Users/oliververity/Library/Logs/Homebrew/kismet/02.make:
(cd capture_bladerf_wiphy && /Applications/Xcode.app/Contents/Developer/usr/bin/make)
clang++ -std=gnu++17 -MM -MP -Wall -Wno-unknown-warning-option -Wno-deprecated-declarations -Wno-format-truncation -Wno-unused-local-typedefs -Wno-unused-function -Wno-infinite-recursion -g -I. -fPIE -g -O2 -O3 -I/opt/homebrew/Cellar/protobuf/21.12/include -DKS_STR_ENCODING_NONE -I/opt/homebrew/Cellar/libpcap/1.10.3/include -I/opt/homebrew/opt/openssl@3/include -I/opt/homebrew/opt/openssl@3/include  -MT log_tools/kismetdb_to_wiglecsv.cc.o log_tools/kismetdb_to_wiglecsv.cc -MF log_tools/kismetdb_to_wiglecsv.cc.d
clang++ -std=gnu++17 -MM -MP -Wall -Wno-unknown-warning-option -Wno-deprecated-declarations -Wno-format-truncation -Wno-unused-local-typedefs -Wno-unused-function -Wno-infinite-recursion -g -I. -fPIE -g -O2 -O3 -I/opt/homebrew/Cellar/protobuf/21.12/include -DKS_STR_ENCODING_NONE -I/opt/homebrew/Cellar/libpcap/1.10.3/include -I/opt/homebrew/opt/openssl@3/include -I/opt/homebrew/opt/openssl@3/include  -MT log_tools/kismetdb_dump_devices.cc.o log_tools/kismetdb_dump_devices.cc -MF log_tools/kismetdb_dump_devices.cc.d
clang++ -std=gnu++17 -MM -MP -Wall -Wno-unknown-warning-option -Wno-deprecated-declarations -Wno-format-truncation -Wno-unused-local-typedefs -Wno-unused-function -Wno-infinite-recursion -g -I. -fPIE -g -O2 -O3 -I/opt/homebrew/Cellar/protobuf/21.12/include -DKS_STR_ENCODING_NONE -I/opt/homebrew/Cellar/libpcap/1.10.3/include -I/opt/homebrew/opt/openssl@3/include -I/opt/homebrew/opt/openssl@3/include  -MT log_tools/kismetdb_statistics.cc.o log_tools/kismetdb_statistics.cc -MF log_tools/kismetdb_statistics.cc.d
clang++ -std=gnu++17 -MM -MP -Wall -Wno-unknown-warning-option -Wno-deprecated-declarations -Wno-format-truncation -Wno-unused-local-typedefs -Wno-unused-function -Wno-infinite-recursion -g -I. -fPIE -g -O2 -O3 -I/opt/homebrew/Cellar/protobuf/21.12/include -DKS_STR_ENCODING_NONE -I/opt/homebrew/Cellar/libpcap/1.10.3/include -I/opt/homebrew/opt/openssl@3/include -I/opt/homebrew/opt/openssl@3/include  -MT log_tools/kismetdb_to_kml.cc.o log_tools/kismetdb_to_kml.cc -MF log_tools/kismetdb_to_kml.cc.d
clang++ -std=gnu++17 -MM -MP -Wall -Wno-unknown-warning-option -Wno-deprecated-declarations -Wno-format-truncation -Wno-unused-local-typedefs -Wno-unused-function -Wno-infinite-recursion -g -I. -fPIE -g -O2 -O3 -I/opt/homebrew/Cellar/protobuf/21.12/include -DKS_STR_ENCODING_NONE -I/opt/homebrew/Cellar/libpcap/1.10.3/include -I/opt/homebrew/opt/openssl@3/include -I/opt/homebrew/opt/openssl@3/include  -MT log_tools/kismetdb_to_gpx.cc.o log_tools/kismetdb_to_gpx.cc -MF log_tools/kismetdb_to_gpx.cc.d
clang++ -std=gnu++17 -MM -MP -Wall -Wno-unknown-warning-option -Wno-deprecated-declarations -Wno-format-truncation -Wno-unused-local-typedefs -Wno-unused-function -Wno-infinite-recursion -g -I. -fPIE -g -O2 -O3 -I/opt/homebrew/Cellar/protobuf/21.12/include -DKS_STR_ENCODING_NONE -I/opt/homebrew/Cellar/libpcap/1.10.3/include -I/opt/homebrew/opt/openssl@3/include -I/opt/homebrew/opt/openssl@3/include  -MT log_tools/kismetdb_clean.cc.o log_tools/kismetdb_clean.cc -MF log_tools/kismetdb_clean.cc.d
clang -MM -Wall -Wno-unknown-warning-option -Wno-format-truncation -Wno-unused-function -g -I. -fPIE   -I/opt/homebrew/opt/openssl@3/include -pthread -I/opt/homebrew/Cellar/protobuf-c/1.4.1_1/include -I/opt/homebrew/Cellar/libpcap/1.10.3/include  -I/opt/homebrew/opt/openssl@3/include -I/opt/homebrew/opt/openssl@3/include  -I/opt/homebrew/opt/openssl@3/include  ../wifi_ht_channels.c | sed -e "s/\.o/\.c.o/" > ../wifi_ht_channels.c.d
clang -MM -Wall -Wno-unknown-warning-option -Wno-format-truncation -Wno-unused-function -g -I. -fPIE   -I/opt/homebrew/opt/openssl@3/include -pthread -I/opt/homebrew/Cellar/protobuf-c/1.4.1_1/include -I/opt/homebrew/Cellar/libpcap/1.10.3/include  -I/opt/homebrew/opt/openssl@3/include -I/opt/homebrew/opt/openssl@3/include  -I/opt/homebrew/opt/openssl@3/include  capture_bladerf_wiphy.c | sed -e "s/\.o/\.c.o/" > capture_bladerf_wiphy.c.d
make[2]: *** No rule to make target `../libkismetdatasource.a', needed by `kismet_cap_bladerf_wiphy'.  Stop.
make[2]: *** Waiting for unfinished jobs....
clang++ -std=gnu++17 -MM -MP -Wall -Wno-unknown-warning-option -Wno-deprecated-declarations -Wno-format-truncation -Wno-unused-local-typedefs -Wno-unused-function -Wno-infinite-recursion -g -I. -fPIE -g -O2 -O3 -I/opt/homebrew/Cellar/protobuf/21.12/include -DKS_STR_ENCODING_NONE -I/opt/homebrew/Cellar/libpcap/1.10.3/include -I/opt/homebrew/opt/openssl@3/include -I/opt/homebrew/opt/openssl@3/include  -MT log_tools/kismetdb_to_pcap.cc.o log_tools/kismetdb_to_pcap.cc -MF log_tools/kismetdb_to_pcap.cc.d
make[1]: *** [capture_bladerf_wiphy/kismet_cap_bladerf_wiphy] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [install] Error 2

Do not report this issue to Homebrew/brew or Homebrew/core!`

israelshirk, 2023-01-29:
Same here.  Looking into it but not quite sure what to do.  On BladeRF as well.

israelshirk, 2023-01-29:
Switched to `brew install --HEAD kismet-git` on OSX 13.3 development branch.  Works well till they carve out a release.