kismetwireless/kismet #467: libprotobuf23 not installable

colorpride, 2023-04-02:
OS: Debian Bookworm

Error:
kismet-core : Depends:  libprobuf23 but it is not installable 

sudo apt search libprobut*
libprobut32 / testing

Thanks.

colorpride, 2023-06-13:
do you have news?

jamieb999, 2023-06-21:
I  am getting the same error message on debian 12 (Bookworm) as well

uname -a
Linux laptop 6.1.0-9-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.27-1 (2023-05-08) x86_64 GNU/Linux

/etc/apt/sources.list.d# cat kismet.list
deb [signed-by=/usr/share/keyrings/kismet-archive-keyring.gpg] https://www.kismetwireless.net/repos/apt/git/bookworm bookworm main

apt-get install kismet
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 kismet-core : Depends: libprotobuf23 but it is not installable
               Recommends: kismet-capture-ti-cc-2540 but it is not going to be installed
E: Unable to correct problems, you have held broken packages.

apt-cache search ibprotobuf
libprotobuf-dev - protocol buffers C++ library (development files) and proto files
libprotobuf-java - Java bindings for protocol buffers
libprotobuf-lite32 - protocol buffers C++ library (lite version)
libprotobuf32 - protocol buffers C++ library
libprotobuf-c-dev - Protocol Buffers C static library and headers (protobuf-c)
libprotobuf-c1 - Protocol Buffers C shared library (protobuf-c)
protobuf-c-compiler - Protocol Buffers C compiler (protobuf-c)
libprotobuf-java-format-java - Library to serialize protobuf messages to XML, JSON and HTML
libprotobuf2-java - Java bindings for protocol buffers

There is no  libprotobuf23  in Debian 12 (Bookworm) which is now stable. 
Closest we have in Debian 12 is "libprotobuf32 - protocol buffers C++ library"

Please fix this! 

I see another person is having the same issue that I am having. 

It is impossible to get Kismet to install on debian 12 due to the 
unmet dependencies issue. 

Thank you,

Jamie (she / her)

jamieb999, 2023-06-21:
A temporary solution seems to be this on Debian 12

Downloada copy of   libprotobuf23_3.12.4-1_amd64.deb
from https://packages.debian.org/bullseye/amd64/libprotobuf23/download

https://packages.debian.org/bullseye/libprotobuf23

This is from Bullseye but works with Bookworm still as of 
June 21 2023 this may change in the future. Do this at
your own risk! 

Go to where the .deb file just downloaded to 
Then as root run dpkg  -i  libprotobuf23_3.12.4-1_amd64.deb

After that then do an
 
apt-get install kismet

after I installed libprotobuf23_3.12.4-1_amd64.deb 
then Kismet installed. 

I hope this temporary solution helps other people
out there too.

Jamie (she / her)

davidcc73, 2025-03-29:
Still works by 29/03/2025