kismetwireless/kismet #468: arm-linux-gnueabihf/ld-2.31.so

JoeTester1965, 2023-04-15:
Getting this error intermittently on raspberry a Pi 4 running bullseye, kismet built from source, using a TPLINK TL-WN722N for  WiFi and Ubertooth one for BLE:

Object "/usr/lib/arm-linux-gnueabihf/ld-2.31.so, at 0xf718090f, in
#0  | Source "/home/pi/Play/kismet/backward.h", line 2046, in load_from
    |  2044: #endif
    |  2045:            if (error_addr) {
    | >2046:                    st.load_from(error_addr, 32);
    |  2047:            } else {
    |  2048:                    st.load_here(32);
      Source "/home/pi/Play/kismet/backward.h", line 631, in sig_handler [0x82b413]
        628:            return size();
        629:    }
        630:    size_t load_from(void* addr, size_t depth=32) {
      > 631:            load_here(depth + 8);
        632:
        633:            for (size_t i = 0; i < _stacktrace.size(); ++i) {
        634:                    if (_stacktrace[i] == addr) {
Bus error (Invalid address alignment [0xf408f2c5])
FATAL:  Error during write(): Broken pipe
FATAL:  Error during write(): Broken pipe

dragorn, 2023-04-15:
Unfortunately that's inside the error handler so it doesn't reveal too much. If you can get it to happen even semi-regularly, if you could try running it w the debug instructions from https://kismetwireless.net/docs/dev/debugging/ ... you probably won't want to compile it yourself w/ debugging on a pi, but if you pick up around step 4 and run 'gdb /usr/bin/kismet', then follow the rest of the steps to get a log when it crashes, that might help.

If you're not running the nightly kismet packages, give those a try, too. A new release is fairly close, and they fix a lot of other bugs, so there's a chance it's already been handled.

tnorman42, 2023-06-28:
I just ran into this also on a fresh build from master on my RPi 4 running Raspberry Pi OS. Here's the backtrace.

#0  0x0028ae78 in crc32_16bytes (data=0xf5521522, length=<optimized out>, previousCrc32=previousCrc32@entry=0) at crc32.cc:429
#1  0x0028b89c in crc32_fast (data=<optimized out>, length=<optimized out>, previousCrc32=previousCrc32@entry=0) at crc32.cc:545
#2  0x00548998 in operator() (__closure=0xa315d4, in_pack=<error reading variable: Cannot access memory at address 0x9f310d06>)
    at packetchain.cc:210
#3  0x005496c0 in std::__invoke_impl<int, packet_chain::packet_chain()::<lambda(std::shared_ptr<kis_packet>)>&, std::shared_ptr<kis_packet> > (__f=...) at /usr/include/c++/10/bits/invoke.h:60
#4  std::__invoke_r<int, packet_chain::packet_chain()::<lambda(std::shared_ptr<kis_packet>)>&, std::shared_ptr<kis_packet> > (
    __fn=...) at /usr/include/c++/10/bits/invoke.h:113
#5  std::_Function_handler<int(std::shared_ptr<kis_packet>), packet_chain::packet_chain()::<lambda(std::shared_ptr<kis_packet>)> >::_M_invoke(const std::_Any_data &, std::shared_ptr<kis_packet> &&) (__functor=..., __args#0=...)
    at /usr/include/c++/10/bits/std_function.h:291
#6  0x0053f050 in std::function<int (std::shared_ptr<kis_packet>)>::operator()(std::shared_ptr<kis_packet>) const (
    __args#0=std::shared_ptr<kis_packet> (empty) = {...}, this=<optimized out>) at /usr/include/c++/10/bits/std_function.h:622
#7  packet_chain::packet_queue_processor (this=0xa39980, packet_queue=<optimized out>) at packetchain.cc:454
#8  0x00540de4 in operator() (__closure=0xbf14c4) at packetchain.cc:316
#9  std::__invoke_impl<void, packet_chain::start_processing()::<lambda()> > (__f=...) at /usr/include/c++/10/bits/invoke.h:60
#10 std::__invoke<packet_chain::start_processing()::<lambda()> > (__fn=...) at /usr/include/c++/10/bits/invoke.h:95
#11 std::thread::_Invoker<std::tuple<packet_chain::start_processing()::<lambda()> > >::_M_invoke<0> (this=0xbf14c4)
    at /usr/include/c++/10/thread:264
#12 std::thread::_Invoker<std::tuple<packet_chain::start_processing()::<lambda()> > >::operator() (this=0xbf14c4)
    at /usr/include/c++/10/thread:271
#13 std::thread::_State_impl<std::thread::_Invoker<std::tuple<packet_chain::start_processing()::<lambda()> > > >::_M_run(void) (
    this=0xbf14c0) at /usr/include/c++/10/thread:215
#14 0xf7bdb150 in ?? () from /lib/arm-linux-gnueabihf/libstdc++.so.6
#15 0xf7aba310 in start_thread (arg=0xf2aff300) at pthread_create.c:477
#16 0xf74d1da8 in ?? () at ../sysdeps/unix/sysv/linux/arm/clone.S:73 from /lib/arm-linux-gnueabihf/libc.so.6
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

kismetwireless, 2023-07-01:
Possibly fixed in the latest builds.

tnorman42, 2023-07-30:
Finally got around to rebuilding with the latest build, and it seems fixed, thanks! Has been running for >1 hour without issue so far.

RFDeck, 2026-11-07:
The backtrace points at crc32_16bytes(), which reads packet data through 32-bit pointers that aren't always aligned. On 32-bit ARM that can raise a bus error. Since 0c1b4e29 (2023-07-01), 32-bit builds use the bytewise CRC instead (crc32.h), and the last comment here confirms it ran without errors after rebuilding. That change is in every release since 2023-07-R1. Can this be closed?

JoeTester1965, 2026-11-08:
> requests on the go with GitHub Mobile for iOS
> and Android
> Download it today!

Yes thanks for that