kismetwireless/kismet #471: Random crashing on Pi 4

JoeTester1965, 2023-05-19:
Rebuilt with latest code here.

Crash seems to happen randomly after detecting some new 802.11 WiFi devices e.g,.:

INFO: Detected new 802.11 Wi-Fi device C8:3A:35:22:B1:F0
Stack trace (most recent call last) in thread 12510:
#1    Object "/usr/lib/arm-linux-gnueabihf/ld-2.31.so, at 0xf702c90f, in
#0  | Source "/home/pi/Play/kismet/backward.h", line 2046, in load_from
    |  2044: #endif
    |  2045:            if (error_addr) {
    | >2046:                    st.load_from(error_addr, 32);
    |  2047:            } else {
    |  2048:                    st.load_here(32);
      Source "/home/pi/Play/kismet/backward.h", line 631, in sig_handler [0x8bc6cb]
        628:            return size();
        629:    }
        630:    size_t load_from(void* addr, size_t depth=32) {
      > 631:            load_here(depth + 8);
        632:
        633:            for (size_t i = 0; i < _stacktrace.size(); ++i) {
        634:                    if (_stacktrace[i] == addr) {
Bus error (I

Have attached gdb.txt 
[gdb.txt](https://github.com/kismetwireless/kismet/files/11516462/gdb.txt)

JoeTester1965, 2023-05-20:
Afraid that did not fix it, let me know if you want another gdb output.

JoeTester1965, 2023-05-20:
Have attached a new gdb log in case you need:
 
[gdb.txt](https://github.com/kismetwireless/kismet/files/11522322/gdb.txt)

dragorn, 2023-05-20:
I’ll need more gdb, yeah.

JoeTester1965, 2023-05-22:
I have attached gdb for debugging. FYI this problem also occurs in the latest official buster package.
[gdb.txt](https://github.com/kismetwireless/kismet/files/11529906/gdb.txt)

kismetwireless, 2023-05-22:
Unfortunately that doesn't give me any further insights; all I can suggest is trying the latest git (there have been many changes since last week), or try a different platform.  I can't recreate the crash here, so I'm stuck unless there's some way to get more info than what is shown in the traces.

JoeTester1965, 2023-05-27:
FYI this bug is only apparent on the latest Raspbian release Bullseye , does not seem to happen on Buster.

closed 2023-05-27 by JoeTester1965