kismetwireless/kismet #478: Building with flag --enable-wifi-coconut leads to errors of unknown type name

sarcasticadmin, 2023-08-15:
## Description

Attempting to build off latest `master` (`f5a3205895bd013052403b30a972719cb16c6119`). Building with configure flags: `--prefix=/nix/store/v019cx1hjm904jdwbg0478in6svm6dg1-kismet-2023-08-11_unstable --disable-libnm --disable-lmsensors --enable-wifi-coconut` was resulting in the following error:
```
gcc -rdynamic -o kismet_cap_ti_cc_2540 capture_ti_cc_2540.c.o  ../libkismetdatasource.a -lcap -L/nix/store/z3lw8cy01hln5fh55vnvpb92wwa2kcga-libwebsockets-4.3.2/lib -lwebsockets -lpthread -ore/zr3y0lyxl7yq7m9nny5bybq9drr2m5ri-protobuf-c-1.4.1/lib -lprotobuf-c -lm -L/nix/store/rij1if80f4hp1vl7lfbi37msra8yl84n-libusb-1.0.26/lib -lusb-1.0
In file included from libwifiuserspace/kernel/nl80211.h:36,
                 from libwifiuserspace/userspace/userspace.h:19,
                 from capture_hak5_coconut.c:55:
libwifiuserspace/kernel/types.h:40:9: error: unknown type name '__u16'
   40 | typedef __u16 __le16;
      |         ^~~~~
libwifiuserspace/kernel/types.h:41:9: error: unknown type name '__u32'
   41 | typedef __u32 __le32;
      |         ^~~~~
libwifiuserspace/kernel/types.h:42:9: error: unknown type name '__u64'
   42 | typedef __u64 __le64;
      |         ^~~~~
libwifiuserspace/kernel/types.h:44:9: error: unknown type name '__u16'
   44 | typedef __u16 __be16;
      |         ^~~~~
libwifiuserspace/kernel/types.h:45:9: error: unknown type name '__u32'
   45 | typedef __u32 __be32;
      |         ^~~~~
libwifiuserspace/kernel/types.h:46:9: error: unknown type name '__u64'
   46 | typedef __u64 __be64;
      |         ^~~~~
In file included from libwifiuserspace/userspace/userspace.h:21:
libwifiuserspace/kernel/usb.h:92:9: error: unknown type name '__u16'
   92 |         __u16           match_flags;
      |         ^~~~~
libwifiuserspace/kernel/usb.h:95:9: error: unknown type name '__u16'
   95 |         __u16           idVendor;
      |         ^~~~~
libwifiuserspace/kernel/usb.h:96:9: error: unknown type name '__u16'
   96 |         __u16           idProduct;
      |         ^~~~~
libwifiuserspace/kernel/usb.h:97:9: error: unknown type name '__u16'
   97 |         __u16           bcdDevice_lo;
      |         ^~~~~
libwifiuserspace/kernel/usb.h:98:9: error: unknown type name '__u16'
   98 |         __u16           bcdDevice_hi;
      |         ^~~~~
libwifiuserspace/kernel/usb.h:101:9: error: unknown type name '__u8'
  101 |         __u8            bDeviceClass;
      |         ^~~~
libwifiuserspace/kernel/usb.h:102:9: error: unknown type name '__u8'
  102 |         __u8            bDeviceSubClass;
      |         ^~~~
libwifiuserspace/kernel/usb.h:103:9: error: unknown type name '__u8'
  103 |         __u8            bDeviceProtocol;
      |         ^~~~
libwifiuserspace/kernel/usb.h:106:9: error: unknown type name '__u8'
  106 |         __u8            bInterfaceClass;
      |         ^~~~
libwifiuserspace/kernel/usb.h:107:9: error: unknown type name '__u8'
  107 |         __u8            bInterfaceSubClass;
      |         ^~~~
libwifiuserspace/kernel/usb.h:108:9: error: unknown type name '__u8'
  108 |         __u8            bInterfaceProtocol;
      |         ^~~~
libwifiuserspace/kernel/usb.h:111:9: error: unknown type name '__u8'
  111 |         __u8            bInterfaceNumber;
      |         ^~~~
cc1: note: unrecognized command-line option '-Wno-unknown-warning-option' may have been intended to silence earlier diagnostics
make[1]: *** [Makefile:39: capture_hak5_coconut.c.o] Error 1
make[1]: Leaving directory '/build/source/capture_hak5_wifi_coconut'
make: *** [Makefile:388: capture_hak5_wifi_coconut/kismet_cap_hak5_wifi_coconut] Error 2
make: *** Waiting for unfinished jobs....
make[1]: Leaving directory '/build/source/capture_ti_cc_2540'
build failed in buildPhase with exit code 2
```

[full-fail-kismet-build.log](https://github.com/kismetwireless/kismet/files/12341680/full-fail-kismet.log)

## System Info

`NixOS 23.05`:

```
$ uname -a
Linux driver 6.1.36 #1-NixOS SMP PREEMPT_DYNAMIC Wed Jun 28 09:12:41 UTC 2023 x86_64 GNU/Linux
```

```
$ gcc --version
gcc (GCC) 12.2.0
Copyright (C) 2022 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

hughobrien, 2023-10-21:
I've also experienced this with 2023-07-R1

sarcasticadmin, 2023-10-21:
@hughobrien I think theres been some recent work on `master` to fix this: https://github.com/kismetwireless/kismet/commit/82e0ca25d54af40e59fe7f3e08c60ef02c8b51a3

I havent got around to testing it myself yet, but if you want to try it out :slightly_smiling_face:

hughobrien, 2023-10-22:
@sarcasticadmin Thank you, do you think the nixpkg can be updated without Coconut support or is that something you make use of? Suggest we move discussion to the draft PR linked above.

sarcasticadmin, 2023-10-23:
@hughobrien just checked and building with `--enable-wifi-coconut` is still failing for me against latest `master`: https://github.com/kismetwireless/kismet/commit/716dffc65bd39240598aa1e2ec2b0be17c352b32