kismetwireless/kismet #479: Proble with devicefound feature in kismet_alerts.conf

dd121, 2023-08-19:
How do I get alerts with multiple macadresses. Use case is finding specific OUI for example 
devicefound="70:4C:A5:xx:xx:xx,71:4C:A5:xx:xx:xx, etc..."

If I format it this way I dont get a hit. Also not with a mask like devicefound=70:4C:A5:00:00:00 or devicefound=70:4C:A5:::.
I only get a hit if I put in the exact macaddress. like deficefound=70:4C:A5:B5:CC:C1

My use case is to scan for specific wifi devices and I have a list with over 200 Vendor ID's and I want an allert if one is found. 
I was thinking the devicefound alert was the best option, but if someone has a better option I am open to it. An alert is not necessary if there is another option to filter them out.

dragorn, 2023-08-19:
You need to specify the Mac address as a mask, the same as a net mask; there should be examples in the docs but it should just be something like as:bb:cc:00:00:00/ff:ff:ff:00:00:00 to select the oui

dd121, 2023-08-19:
> You need to specify the Mac address as a mask, the same as a net mask; there should be examples in the docs but it should just be something like as:bb:cc:00:00:00/ff:ff:ff:00:00:00 to select the oui
> […](#)

Yes, thank you I just found the same answer in the code. Thanks for the reply!!

closed 2023-08-19 by dd121