kismetwireless/kismet #483: APSpoof alert doesn't trigger.

BumbleB-NL, 2023-09-19:
The APSpoof alert doesn't trigger when a MAC address that is not in the allow list advertises a SSID.

I am using the following config:
`apspoof=NAME:ssid="MYSSID",validmacs="00:11:22:33:44:55"`

There are 4 more AP's advertising the same SSID using a different MAC but they do not trigger an alert.

I am using kismet version 2023-07-R1, I've also tried in a docker container which has the same issue using version 2023-00-git(might not be a real version).

psheneetsakash, 2025-01-17:
hi, were you able to fix it?

BumbleB-NL, 2025-09-03:
Unfortunately I have not been able to find a fix for this issue.

RFDeck, 2026-11-01:
This should be fixed since kismet-2025-09-R1. On builds with libpcre2, the SSID regex of an APSPOOF rule was
compiled, but no match data was ever allocated for it. `pcre2_match()` then failed on every compare, so the rule never
matched. 7bc6cf32 ("pcre2: Fix pcre2 matching not matching", 2024-10-06) allocates the match data. It does the same
for the other PCRE2 users: device view and tracked element regex filters, UAV matching, and the wiglecsv SSID filter.

The 2023-07 releases from the report don't have that fix. Could you retest with kismet-2025-09-R1 or newer? If APSPOOF
still stays quiet, please post the `apspoof=` line, the Kismet version, and whether it was built with libpcre2.