kismetwireless/kismet #486: Manual resizing of panes not possible anymore

fkek19999, 2023-10-11:
Since the new version, the bottom panes (with the messages and channels and plugin tabs) are not resizable anymore. 
Is this with reason or a bug?
If deliberately than I am also interested in a workaround (hacky) to get the possibility to resize these panes to almost full screen.

dragorn, 2023-10-11:
This was deliberate - the datatables jquery entity used for the main display has no reasonable alternative and does not handle resizing panes correctly.

There is a minimize/maximize button in the ui to full screen the primary pane.

-m

fkek19999, 2023-10-11:
I am using the kestrell plugin with a map and I used to enlarge this map in the previous version.  Can you think of a workaround?

d3k4-k3rb3r0s, 2024-04-23:
Windows are also not resizable, resulting in blocked access to certain. Lower parts of windows on tiny pi displays.