kismetwireless/kismet #490: Request: please add support for Raspberry Pi 0 and 2 to the bookworm repos

pdobrien3, 2023-11-12:
I have a raspberry pi 0 and a raspberry pi 2 that I use as a remote capture device.  They work awesome on Bullseye for using just the  Kismet components (kismet-capture-linux-wifi and kismet-capture-linux-bluetooth).  I connect to my main Kismet install using TCP and an Autossh tunnel.  With the recent addition of Bookworm to stable, I would like to update.  I have tried to use the package installers (kali, bookworm, and bullseye) of the 2 component but run into dependencies error.  I have also tried to build my own package locally but it was beyond my abilities.  Thanks in advance for your consideration.

pdobrien3, 2023-11-13:
I struggled through trying to build locally several times yesterday.  Made it all the way to `sudo make suidinstall`.  That step crashes the Pi everytime.  Even tried to put the SD card in a Pi3B+ after a successful `make datasources` and it crashed the Pi 3B+ also.

pdobrien3, 2023-11-13:
Here is the steps I took:

Install dependencies:
`sudo apt install build-essential git libwebsockets-dev pkg-config zlib1g-dev libnl-3-dev libnl-genl-3-dev libcap-dev libpcap-dev libnm-dev libdw-dev libsqlite3-dev libprotobuf-dev libprotobuf-c-dev protobuf-compiler protobuf-c-compiler libsensors4-dev libusb-1.0-0-dev python3 python3-setuptools python3-protobuf python3-requests python3-numpy python3-serial python3-usb python3-dev python3-websockets librtlsdr0 libubertooth-dev libbtbb-dev
`
`sudo apt-get install -y libmosquitto-dev` (initial install)

```
cd ~/downloads
git clone https://www.kismetwireless.net/git/kismet.git 
cd ~/downloads/kismet
./configure --disable-libwebsockets --enable-capture-tools-only
make datasources
sudo make suidinstall
```  (never could make it through this step).

michalpmba, 2024-01-27:
I have successfully setup all capture tools on my raspberry pi 2 with bookworm installed on it. I just build datasources only after cloning kismet repo like this :

1. `apt install build-essential git libwebsockets-dev pkg-config \
zlib1g-dev libnl-3-dev libnl-genl-3-dev libcap-dev libpcap-dev \
libnm-dev libdw-dev libsqlite3-dev libprotobuf-dev libprotobuf-c-dev \
protobuf-compiler protobuf-c-compiler libsensors4-dev libusb-1.0-0-dev \
python3 python3-setuptools python3-protobuf python3-requests \
python3-numpy python3-serial python3-usb python3-dev python3-websockets \
librtlsdr0 libubertooth-dev libbtbb-dev libmosquitto-dev`
2. `./configure --enable-capture-tools-only`
3. `make datasources -j$(nproc)`
4. `make binsuidinstall -j$(nproc)`