kismetwireless/kismet #498: REST-API. "Devices by capture source" returns an error 404

PavelBiz, 2023-12-27:
Hello.
On version 2023-07 when requesting results by source
"/devices/views/seenby-uuid/{UUID}/devices.json"
In response, error 404 is returned. 
On version 2022-08, everything worked correctly.

hobobandy, 2023-12-27:
In git-verison, this endpoint works: /devices/views/seenby-{UUID}/devices.json

Not sure if intended since documented otherwise.

Possible relevent code: devicetracker.cc - line 2018

PavelBiz, 2023-12-28:
This endpoint works: /devices/views/seenby-{UUID}/devices.json
in version 2023-07 and in version 2022-08, 

But for some reason, the documented endpoint has a different type

RFDeck, 2026-11-01:
The 404 comes from how the alternate view paths are registered. A view built with path parts like
`{"seenby-uuid", <uuid>}` joins them with a slash after each part, and `register_urls()` adds another before `devices`.
The route ends up as `/devices/views/seenby-uuid/<UUID>//devices`, which neither the documented path nor the one with
`//` reaches. The same happens to `/devices/views/phy/<phy>/devices`. This has been in `devicetracker_view.cc` since at
least 2022-08-R1. `seenby-<UUID>` is the view id, which is why that path works.

#696 joins the parts with a slash between them. With it, `/devices/views/seenby-uuid/<UUID>/devices.json` answers 200
with the same devices as `seenby-<UUID>`. The UUID has to be written the way Kismet prints it, in upper case, for
either path.