kismetwireless/kismet #508: build issue OpenBSD snapshot

buzzdeee, 2024-02-25:
Hi,

since all work on the OpenBSD WiFi support, I in the meantime upgraded my host, and found that build broke:

```
c++ -std=gnu++17 -Wall -Wno-unknown-warning-option -Wno-deprecated-declarations -Wno-format-truncation -Wno-unused-local-typedefs -Wno-unused-function -Wno-infinite-recursion -g -I. -fPIE -O2 -pipe -g -O0 -O3 -I/usr/local/include -DPROTOBUF_USE_DLLS -Wno-float-conversion -Wno-implicit-float-conversion -Wno-implicit-int-float-conversion -Wno-unknown-warning-option -DNOMINMAX -DKS_STR_ENCODING_NONE   -I/usr/X11R6/include -I/usr/local/include  -c trackedelement.cc -o trackedelement.cc.o
In file included from trackedelement.cc:26:
In file included from ./trackedelement.h:42:
In file included from ./globalregistry.h:33:
./robin_hood.h:1445:33: warning: builtin __has_trivial_copy is deprecated; use __is_trivially_copyable instead [-Wdeprecated-builtins]
        Cloner<Table, IsFlat && ROBIN_HOOD_IS_TRIVIALLY_COPYABLE(Node)>()(o, *this);
                                ^
./robin_hood.h:210:51: note: expanded from macro 'ROBIN_HOOD_IS_TRIVIALLY_COPYABLE'
#    define ROBIN_HOOD_IS_TRIVIALLY_COPYABLE(...) __has_trivial_copy(__VA_ARGS__)
                                                  ^
In file included from trackedelement.cc:32:
./alphanum.hpp:307:38: error: no template named 'binary_function' in namespace 'std'; did you mean '__binary_function'?
  struct alphanum_less : public std::binary_function<Ty, Ty, bool>
                                ~~~~~^~~~~~~~~~~~~~~
                                     __binary_function
/usr/include/c++/v1/__functional/binary_function.h:49:1: note: '__binary_function' declared here
using __binary_function = __binary_function_keep_layout_base<_Arg1, _Arg2, _Result>;
^
1 warning and 1 error generated.
gmake: *** [Makefile:838: trackedelement.cc.o] Error 1
*** Error 2 in . (/home/ports/infrastructure/mk/bsd.port.mk:3059 '/home/ports/pobj/amd64/kismet-202307R1/.build_done': @cd /home/ports/pobj/...)
*** Error 2 in /home/ports/mystuff/net/kismet (/home/ports/infrastructure/mk/bsd.port.mk:2702 'all': @lock=kismet-202307R1;  export _LOCKS_H...)
```

not sure, if this is llvm/clang update in the system, or some other dependency i.e. boost.

prior where it was working:
boost-1.80.0
clang version 13.0.0

now not working anymore with these versions:
boost-1.84.0
clang version 16.0.6

my C++ knowledge is rather weak, so don't know where this comes from. Some research lead me to this:
https://stackoverflow.com/questions/33114656/replacement-for-stdbinary-function
but don't really know how to apply it here.

kismetwireless, 2024-02-26:
Pretty sure this was fixed in git in ~august & will be in the next release once some other work is done; possibly you're on the stable branch still?

commit 5bae9c68ce45940e03fda7d8abbbfda0fca9e95c
Author: Mike Kershaw / Dragorn ***@***.***>
Date: Sun Aug 6 16:55:48 2023 -0400

alphanum: Remove usage of std::binary_function in the function since
it's been long deprecated and is beginning to cause problems with newer compilers

buzzdeee, 2024-02-26:
yuck, indeed, I was a number of times in the wrong directory when starting the build, sorry for the noise.

closed 2024-02-26 by buzzdeee