kismetwireless/kismet #514: Kismet not recording Bluetooth devices in wiglecsv logs

Corvus4n6, 2024-03-27:
I found this issue first with 2023-07.R1 and also under 9999+git2024-03-27r0e3b7a50-1 running under Debian Bookworm. 

Observed Bluetooth devices are not logged to the wiglecsv file and are not exported using kismetdb_to_wiglecsv.

Reviewing the Kismet database file, I can see the devices listed in the tables data and devices, but there is no GPS data associated with these devices despite confirmation that GPS is working correctly. GPS data is logged for Bluetooth devices in the messages table, with the lat and lon, when each new device is detected, also confirming that GPS is enabled and functioning.

I have been sure to have enabled all logging, including allowing logging of BTLE and 'random' devices, but the data does not record to the wiglecsv file. I assume this is an issue with the lack of packet data from BT detections, but I hope the fix will be an easy one.

Thank you for your hard work creating and maintaining this project that gets me out of the house to go on warwalks. :smile:

k-rku, 2024-04-03:
Have noticed same issue. Haven't tried other versions yet though.
Running 2023-07-R1 on a Pi4b and the wiglecsv files don't contain any bluetooth devices.

ODenssi, 2024-04-03:
I'm also experiencing the same issue. Running 2023-07-R1 on a Pi4b too and I never got it to log bluetooth devices. Discovering of the BLTE devices works fine.

k-rku, 2024-04-05:
Today I removed 2023-07-R1 and installed the GIT version.
This also doesn't write BTLE info to the wiglecsv.

ODenssi, 2024-04-21:
I also tried this with the Debian Bullseye 32bit version. Still not writing bluetooth to wiglecsv.

nopthetop, 2024-05-27:
Confirming that I am also experiencing this issue with `Kismet 2023-07-R1`.  I have a sdr433 interface that is logging and longitude and latitude are properly logging for those as seen in the devices table.

ODenssi, 2024-08-07:
This issue also persists with .kml logs, atleast when converting with kismetdb_to_* tool. I'm using hci0 interface for recoding bluetooth.

ODenssi, 2024-08-29:
Git release version received a fix on 'datasource_linux_bluetooth.cc' couple of weeks ago to address this issue. Unfortunately the issue still persists for me. Has anyone had any luck with the fix?

peterlavelle, 2024-10-04:
Getting the same result from a git checkout of master yesterday on Raspberry Pi OS Bullseye, 32 bit

Using the raspberrypi 5.15.44-Re4son-v7+ kernel on here, fwiw.

mickdupreez, 2025-02-24:
Same here problem for me

designer2k2, 2025-05-10:
The issue is still there with 2023-07.R2 for me

FadeVT, 2025-06-02:
I am experiencing the same issue.

RFDeck, 2026-11-01:
This was fixed in February 2026 and is in kismet-2026-10-R1:

- 931e74d0 (2026-02-02): the live wiglecsv log writes Bluetooth and BTLE devices.
- 6e69e21b (2026-02-06): `kismetdb_to_wiglecsv` reads Bluetooth and BTLE devices from the `data` table, with signal
  levels.
- 9a8c3ef7 (2026-02-07): `kismetdb_to_wiglecsv` warns when a kismetdb is too old to hold them.

`kismetdb_to_wiglecsv` needs a log written by one of these versions (kismetdb version 10). Older logs, like the ones in
this thread, don't have the Bluetooth records it reads, so it skips Bluetooth for them and says so. Could you retest
with kismet-2026-10-R1 and a new log?