kismetwireless/kismet #515: Kismet not recognizing GPSD data

OnyxDash, 2024-03-29:
I recently started using Kismet Wireless on my Raspberry Pi 5, and everything works fine excluding the GPS. When I first tested it using my MediaTek wifi card and u-blox 7 GPS dongle, it connected fine. Once I set up a script to automatically launch Kismet on startup (since I don't have a monitor for my Pi in my car), the GPS doesn't connect, and hangs with the error "No readable data from GPSD in the last 30 seconds, reconnecting."

The startup script contains the following:
sudo airmon-ng start wlan1
sudo gpsd /dev/ttyACM0
kismet -c wlan1mon

It's very possible there's something I'm missing, but in case I'm not I wanted to make an issue report.

k-rku, 2024-04-04:
I also had multiple issues with GPSD 3.22.
After updating it to 3.25, things were working for me.

These are the steps I did to update to 3.25.
=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=
Stop possibly interfering gpsd services:
------------------------------------------------
sudo systemctl stop gpsd.socket
sudo systemctl disable gpsd.socket
------------------------------------------------

Here's the subsequent install:
------------------------------------------------
sudo apt install -y scons \
  libncurses-dev python-dev-is-python3 pps-tools git-core asciidoctor python3-matplotlib\
  build-essential manpages-dev pkg-config python3-distutils

wget http://download.savannah.gnu.org/releases/gpsd/gpsd-3.25.tar.gz
tar -xzf gpsd-3.25.tar.gz

cd gpsd-3.25/

sudo scons
sudo scons udev-install
sudo reboot
------------------------------------------------

ODenssi, 2024-04-04:
This step helped me with gpsd issues: 
"sudo nano /etc/ default/gpsd"
--> ADD: DEVICES="/dev/ttyACM0"
"/dev/ttyACM0" is where your GPS is mounted. It may be different for you. "ttyACM0" is default for bookworm in my experience. 
Also ADD: GPSD_OPTIONS="-n -b".

v-wan, 2024-05-03:
I was spending a whole two days trying to figure out why Kismet wasnt showing GPS data in that upper left corner, even downloaded the newest gpsd and used the kismet-2023-07 branch instead of master... turns out I've setup everything up correctly, but the GPS module wasn't receiving signals since it was under a roof. After exposing it to the sky it worked like a charm!