kismetwireless/kismet #516: Current Kismet susceptible to a number of security issues?

robvandenbrink, 2024-04-04:
I just downloaded the latest version of Kismet, I really like the new look, especially the use of the browser as the client now.  However, I'm seeing some older jquery comnponents that introduce some security issues:

jquery 3.1.0 and jquery-3.1.0.min , susceptible to CVE-2019-11358, CVE-2020-11023, CVE-2020-11022
jquery-ui.min, susceptible to a number of XSS vulnerabilities: CVE-2021-41182, CVE-2021-41184, CVE-2021-41183, CVE-2022-31160

I'm also able to inject a CRLF into the KISMET cookie, allowing an injection of a new response header.  I'm still looking if that allows me to send a split response / request smuggling attack

happy to chat if you need more info: rob@coherentsecurity.com

dragorn, 2024-04-04:
Thanks for the heads up; I'll drop the latest jquery into the git version and see what it breaks.

Since Kismet isn't a traditional multi-user web app (there's no path for multiple logins or cross-login communication) there isn't much opportunity to inject to anyone but yourself - there's also not really any router system where parts of the web ui are exposed as urls.

Obviously leaving libraries with known issues in place is no good, and it looks like the latest jquery can just be dropped in without breaking anything obvious.

I'd encourage testing with the nightly builds, as there has been a lot of changes since the last release; the drop-in of the new jquery will hit the build server tonight and show up in tomorrows nightly packages.

There isn't a hard ETA for a new release yet, but once some other fixes and new device support is done it will likely be time & will roll in the new js.

An easier way to reach me more interactively, if I'm wrong about the implications even for a single-user system, is the discord server.

-m

robvandenbrink, 2024-04-04:
Thanks, this was just a heads-up
Your experience of "upgrade jquery and see what breaks" completely mirrors mine, enjoy that journey I guess :-)
Great tool!

closed 2024-04-18 by kismetwireless