kismetwireless/kismet #518: Build fails with strict-aliasing violations

eli-schwartz, 2024-04-05:
I tried to build with the following *FLAGS to optimize the build: `-flto=4 -Werror=odr -Werror=lto-type-mismatch -Werror=strict-aliasing`

The -Werror=* flags are important to detect cases where the compiler can try to optimize based on assuming UB cannot happen, and miscompile code that has UB in it. strict-aliasing issues are always bad but LTO can make them even worse.

I got this error:
```
x86_64-pc-linux-gnu-g++ -std=gnu++17 -march=native -fstack-protector-all -O2 -pipe -fdiagnostics-color=always -frecord-gcc-switches -flto=4 -Werror=odr -Werror=lto-type-mismatch -Werror=strict-aliasing  -Wformat -Werror=format-security -Wall -Wno-deprecated-declarations -Wno-format-truncation -Wno-unused-local-typedefs -Wno-unused-function -Wno-infinite-recursion -g -I. -fPIE -march=native -fstack-protector-all  -pipe -fdiagnostics-color=always -frecord-gcc-switches -flto=4 -Werror=odr -Werror=lto-type-mismatch -Werror=strict-aliasing -Wformat -Werror=format-security   -DKS_STR_ENCODING_NONE -I/usr/include/dbus-1.0 -I/usr/lib64/dbus-1.0/include   -c kaitaistream.cc -o kaitaistream.cc.o
kaitaistream.cc: In member function ‘float kaitai::kstream::read_f4be()’:
kaitaistream.cc:256:38: error: dereferencing type-punned pointer will break strict-aliasing rules [-Werror=strict-aliasing]
  256 |     return reinterpret_cast<float &>(t);
      |                                      ^
kaitaistream.cc: In member function ‘double kaitai::kstream::read_f8be()’:
kaitaistream.cc:265:39: error: dereferencing type-punned pointer will break strict-aliasing rules [-Werror=strict-aliasing]
  265 |     return reinterpret_cast<double &>(t);
      |                                       ^
kaitaistream.cc: In member function ‘float kaitai::kstream::read_f4le()’:
kaitaistream.cc:278:38: error: dereferencing type-punned pointer will break strict-aliasing rules [-Werror=strict-aliasing]
  278 |     return reinterpret_cast<float &>(t);
      |                                      ^
kaitaistream.cc: In member function ‘double kaitai::kstream::read_f8le()’:
kaitaistream.cc:287:39: error: dereferencing type-punned pointer will break strict-aliasing rules [-Werror=strict-aliasing]
  287 |     return reinterpret_cast<double &>(t);
      |                                       ^
cc1plus: some warnings being treated as errors
make: *** [Makefile:822: kaitaistream.cc.o] Error 1
```

Downstream report: https://bugs.gentoo.org/877761
Full build log: [build.log](https://github.com/kismetwireless/kismet/files/14880312/build.log)

generalmimon, 2024-07-22:
FYI, this has been fixed in our upstream repo https://github.com/kaitai-io/kaitai_struct_cpp_stl_runtime (https://github.com/kaitai-io/kaitai_struct_cpp_stl_runtime/pull/73), so you may want to update your copy.

RFDeck, 2026-11-06:
The four float readers from the log (`read_f4be`, `read_f8be`, `read_f4le`, `read_f8le`) no longer use
`reinterpret_cast<float &>` / `<double &>`. The Kaitai runtime update in 85a68937 (2025-02-28, "kaitai: merge latest
kaitai stream code") switched them to `bit_cast`, which copies with `memcpy`. That update shipped in 2025-09-R1 and is
in 2026-10-R1.

I didn't build with `-flto -Werror=strict-aliasing` myself, so other spots could still warn under those flags. Could
you check whether 2026-10-R1 builds for you? If it does, this can be closed.