kismetwireless/kismet #519: Hidden / cloaked SSIDs

robvandenbrink, 2024-04-05:
Active SSIDs that aren't broadcasted (ie hidden / cloaked depending on the vendor's taste in poetry) don't show up in the SSID list in the current (web) version as it has in previous (fat client) versions, also not in the devices pane.
Clients of the cloaked SSIDs do however show up in the devices pane, and you can drill down from there.  The SSID still doesn't show up in that path of exploration either though.

Is this by design, or is this maybe a problem in Ubuntu 22.04.4 LTS ?  I'm using the same machine and USB NIC as when I worked with the previous versions

dragorn, 2024-04-05:
On Friday, April 5th, 2024 at 4:50 PM, Rob VandenBrink ***@***.***> wrote:

> Active SSIDs that aren't broadcasted (ie hidden / cloaked depending on the vendor's taste in poetry) don't show up in the SSID list in the current (web) version as it has in previous (fat client) [...]

Access points will show up in the device list as an AP, with broadcasting ssid records including the empty SSID and indicated length, if any. The default device name will be the BSSID of the AP if there is no SSID to derive a name.

Advertised SSIDs
SSID: Unknown
SSID Cloaked / Empty (0 spaces)
Encryption Open
MFP Unavailable
Channel 36HT Mode HT20

Hidden SSIDs will show in the SSID tab per indicated length; depending on the implementation of cloaking they advertise as length 0, length of the original ssid but no content, or length 32.

robvandenbrink, 2024-04-05:
Got it!
It does show up as you indicate, but only after a device associates to it.  This is as you'd expect because there's no info broadcasted, kismet can't see that info until it catches a mgt frame.  
All good, this was just me being dumb

Thanks again for such a quick reply on a Friday!