kismetwireless/kismet #521: Error installing Kismet on M3 Mac pro with Sonoma 14.4

puzzledyk, 2024-04-17:
I get the following error when running brew install --HEAD kismet-git or any other command to install Kismet. I have protobuf and protobuf-c installed at latest versions. 

~ % ./configure --with-protobuf=/opt/homebrew/Cellar/protobuf/26.1 --with-protobuf-c=/opt/homebrew/Cellar/protobuf-c/1.5.0_4

zsh: no such file or directory: ./configure
simonhoglund@MBP-LF7CFL73DW-HEST ~ % brew install --HEAD kismet-git
==> Downloading https://formulae.brew.sh/api/formula.jws.json

==> Downloading https://formulae.brew.sh/api/cask.jws.json

==> Fetching kismetwireless/kismet/kismet-git
==> Cloning https://www.kismetwireless.net/git/kismet.git
Updating /Users/simonhoglund/Library/Caches/Homebrew/kismet-git--git
==> Checking out branch master
Already on 'master'
Your branch is up to date with 'origin/master'.
HEAD is now at 9bf1bc7ce zigbee: merge fix from bergera to populate ticc2531 802.15.4tap header length zigbee: Set original packet lengths
==> Installing kismet-git from kismetwireless/kismet
==> ./configure --disable-silent-rules --enable-wifi-coconut --enable-bladerf --disable-python-tools --with-openssl=/opt/homebrew/opt/openssl@3
==> make install
Last 15 lines from /Users/simonhoglund/Library/Logs/Homebrew/kismet-git/02.make:
make[1]: *** [protobuf_c/kismet.pb-c.h] Error 1
make[1]: *** Waiting for unfinished jobs....
WARNING: All log messages before absl::InitializeLog() is called are written to STDERR
E0000 00:00:1713381153.397546  198852 command_line_interface.cc:1534] Built-in generator --c_out specifies a maximum edition PROTO3 which is not the protoc maximum 2023.
make[1]: *** [protobuf_c/datasource.pb-c.h] Error 1
WARNING: All log messages before absl::InitializeLog() is called are written to STDERR
E0000 00:00:1713381153.400488  198857 command_line_interface.cc:1534] Built-in generator --c_out specifies a maximum edition PROTO3 which is not the protoc maximum 2023.
make[1]: *** [protobuf_c/linuxbluetooth.pb-c.h] Error 1
WARNING: All log messages before absl::InitializeLog() is called are written to STDERR
E0000 00:00:1713381153.406371  198912 command_line_interface.cc:1534] Built-in generator --c_out specifies a maximum edition PROTO3 which is not the protoc maximum 2023.
make[1]: *** [protobuf_c/eventbus.pb-c.h] Error 1
WARNING: All log messages before absl::InitializeLog() is called are written to STDERR
E0000 00:00:1713381153.432209  198917 command_line_interface.cc:1534] Built-in generator --c_out specifies a maximum edition PROTO3 which is not the protoc maximum 2023.
make[1]: *** [protobuf_c/kismet.pb-c.c] Error 1
make: *** [install] Error 2

kismetwireless, 2024-04-18:
This is a conflict introduced by the new protobuf library, which libprotobuf-c does not handle.  Even the examples in protobuf-c are uncompileable.  

Until libprotobuf-c picks up the changes (and releases a new version, and brew updates to the new version), you will need to install an older version of libprotobufs/protoc.

kismetwireless, 2024-04-19:
Since it looks like we have no idea when pb-c is going to update to work w/ the changes made in protobufs-26.1, i changed the build to package pregenerated intermediary pb-c.c and pb-c.h generations of the protobuf files; pb-c seems to care a lot less about slight variations between the intermediaries and the runtime; this wouldn't work with protobufs itself in C++.

crckrberries, 2024-04-30:
@kismetwireless what's the most recent version that doesn't cause this issue? i believe `brew` has a feature to download different versions of some formulae