kismetwireless/kismet #522: (GPS) Error reading from GPSD connection localhost:2947 - Element not found

v-wan, 2024-05-03:
I've got the GPS coordinates to show up in the upper right corner, but for some reason my logs are being spammed with these error messages everytime it gets new coordinates:
```
INFO: (GPS) Connecting to GPSD on localhost:2947
INFO: (GPS) Connected to gpsd server [::1]:2947
INFO: (GPS) Connected to a JSON-enabled GPSD (3.22), enabling JSON mode
ERROR: (GPS) Error reading from GPSD connection localhost:2947 - Element 
       not found
INFO: (GPS) Connecting to GPSD on localhost:2947
INFO: (GPS) Connected to gpsd server [::1]:2947
INFO: (GPS) Connected to a JSON-enabled GPSD (3.22), enabling JSON mode
ERROR: (GPS) Error reading from GPSD connection localhost:2947 - Element 
       not found
INFO: (GPS) Connecting to GPSD on localhost:2947
INFO: (GPS) Connected to gpsd server [::1]:2947
INFO: (GPS) Connected to a JSON-enabled GPSD (3.22), enabling JSON mode
ERROR: (GPS) Error reading from GPSD connection localhost:2947 - Element 
       not found
```

My /usr/local/etc/kismet.conf:
```
gps=gpsd:host=127.0.0.1,port=2947,reconnect=true
```

My /etc/default/gpsd (gpsd 3.22):
```
DEVICES="/dev/ttyACM0"

# Other options you want to pass to gpsd
GPSD_OPTIONS=""

# Automatically hot add/remove USB GPS devices via gpsdctl
USBAUTO="true"
```

michel-ganguin, 2024-08-23:
Same error here, the GPS coordinates shows up. But it does not get recorded with the new Wi-Fi device event (sqlite kismet contains 0,0 instead of the coordinates).

I'm using a network based source with `gpsd udp://0.0.0.0:12345`, the GPS source spams location data every ~500ms. gpsd with other source is working fine.

ireun, 2025-02-03:
Same here.. any help?

phasedistorter, 2025-02-26:
I have the same issue with kismet and GPSd (v3.25) using gpsdRelay (https://github.com/project-kaat/gpsdRelay).

```
INFO: (GPS) Connecting to GPSD on localhost:2947
INFO: (GPS) Connected to gpsd server [::1]:2947
INFO: (GPS) Connected to a JSON-enabled GPSD (3.25), enabling JSON mode
ERROR: (GPS) Error reading from GPSD connection localhost:2947 - Element 
       not found
INFO: (GPS) Connecting to GPSD on localhost:2947
INFO: (GPS) Connected to gpsd server [::1]:2947
INFO: (GPS) Connected to a JSON-enabled GPSD (3.25), enabling JSON mode
ERROR: (GPS) Error reading from GPSD connection localhost:2947 - Element 
       not found
```

GPSd-Forwarder (https://github.com/tiagoshibata/Android-GPSd-Forwarder) seems be more stable in this scenario.
The error still pops up occasionally, so it's not a perfect solution, sadly.

RFDeck, 2026-11-06:
"Element not found" is the Boost.Asio text for `not_found`. `async_read_until` returns it when gpsd sends a line
longer than the read buffer without a newline in it, for example a SKY report with many satellites. Before
b203e32c (2024-06-04, in 2025-09-R1 and 2026-10-R1), that showed up as this error and a reconnect. Since then the
buffer is 8 KiB and the error is no longer logged.

That handler had a remaining problem: it dropped the buffer but didn't start the next read. After such a line Kismet
read nothing more from gpsd until the 30 second data timeout reconnected. #720 keeps reading and skips the
rest of the long line.

The 0,0 coordinates from the second comment aren't explained by this. Kismet doesn't take a position from a gpsd JSON TPV
when lat or lon is 0. If they still show up on 2026-10-R1, a gpsd log (`gpspipe -w`) from that time would help.