kismetwireless/kismet #524: Bluetooth Packets Missing Location Data

chrismuzyn, 2024-06-03:
Hi, 

We noticed while working with kismet that at some point location tracking during Bluetooth and BTLE packet capture was removed.  We have narrowed down the breaking commit to 9193c08, which appears to take away location from generic packets.

I see device_tracker::update_common_device when passed the UCD_UPDATE_LOCATION appears to fetch location information, and phy_btle.cc and phy_bluetooth.cc both call this function with the appropriate flags, however that information is somehow not making it to the kismetdb logs.

Thanks, 
Chris

neil-m-a, 2024-07-29:
Uncommenting the two lines that were commented out in `gpstracker.cc` in the commit you noted ( https://github.com/kismetwireless/kismet/commit/9193c081b692ed0629f733ea6aa710b417f5e68c) restores location data for Bluetooth when exported to KML. I've minimally tested using the master branch and the latest release tag; not sure if if breaks other stuff that I just haven't run across yet.

chrismuzyn, 2024-08-06:
@neil-m-a thanks for figuring that out!  I will consider opening a pull request, to me it seems like kismet would want to provide the functionality before deprecating the old method of bluetooth location.  With releases being few and far between, building master has become the standard way to deploy this software.

chrismuzyn, 2024-10-04:
This has been fixed somewhere in upstream, it is not in a release as of this date.  Thanks.

closed 2024-10-04 by chrismuzyn