kismetwireless/kismet #531: Field simplification for collections (arrays, vectors)

yurybelevskiy, 2024-09-05:
I am pulling the JSON of devices using Kismet API and I am trying to extract some fields from the collections objects such as:

`"dot11.device": {
     .....
     "dot11.device.advertised_ssid_map: [
             {
                  ......,
                  "dot11.advertisedssid.wps_device_name": "Asus Router",
                  ......
             }
     ]
     .....
}`

I have tried using following string in my "fields" array: "dot11.device/dot11.device.responded_ssid_map/dot11.advertisedssid.wps_device_name" and it didn't produce any result.

## Expected result ##

Provided I put following in my "fields" array:
`"fields": ["dot11.device/dot11.device.responded_ssid_map/dot11.advertisedssid.wps_device_name"]`

I am expecting to get back the following structure:
`"dot11.device": {
     .....
     "dot11.device.advertised_ssid_map: [
             {
                  "dot11.advertisedssid.wps_device_name": "Asus Router",
             },
             {
                  "dot11.advertisedssid.wps_device_name": "Another Router",
             },
             {
                  "dot11.advertisedssid.wps_device_name": "And another Router",
             }
     ]
     .....
}`

Please let me know if that is something Kismet supports. I didn't find any examples or mentions of this functionality in documentation, yet it seems when using regex filtering for fields, something like that would work.

itdojo, 2024-10-03:
Here is a snip of what the API returns from the `/devices/views/phydot11_accesspoints/devices.json` endpoint:

```json
{
    "kismet.device.base.macaddr": "AA:BB:CC:0F:47:D5",
    "dot11.device": {
        "dot11.device.last_bssid": "AA:BB:CC:0F:47:D5",
        "dot11.device.responded_ssid_map": [
            {
                "dot11.advertisedssid.wps_device_name": "Xfinity Wireless Gateway",
            }
        ],
        "dot11.device.last_beacon_timestamp": 1727982007,
        "dot11.device.last_probed_ssid_record": {
            "dot11.probedssid.last_time": 1727981980,
            "dot11.probedssid.ssid": "MYSSID",
        },
        "dot11.device.num_associated_clients": 51,
        "dot11.device.advertised_ssid_map": [
            {
                "dot11.advertisedssid.ssidlen": 7,
                "dot11.advertisedssid.maxrate": 144.400000,
            }
        ],
        "dot11.device.client_disconnects": 1,
    },
    "kismet.device.base.manuf": "Vantiva USA LLC",
}
````

This works (example):
```json
{"fields":["dot11.device/dot11.device.last_probed_ssid_record/dot11.probedssid.ssid"]}
```

This does *not* work (example):
```json
{"fields":["dot11.device/dot11.device.responded_ssid_map/dot11.advertisedssid.wps_device_name"]}
```
> HTTP response is 200 but the API returns keys that have a `0` value.

You are trying to drill down into `dot11.device/dot11.device.responded_ssid_map/dot11.advertisedssid.wps_device_name` and Kismet does not appear to support doing that (I have spent hours trying).  
Speaking in python: Notice that the key (`dot11.device.responded_ssid_map` contains a list which has a single dictionary entry whereas `dot11.device.last_probed_ssid_record` has another dictionary as its value (not a list).)  

Having tried and tried it seems you have to return `dot11.device/dot11.device.responded_ssid_map` (`{"fields":["dot11.device/dot11.device.responded_ssid_map"]}`) and parse it in your app to get to `dot11.advertisedssid.wps_device_name`.

[Kismet's docs do not seem to address this.](https://kismetwireless.net/docs/api/command_param/#field-simplification)

All of this is no help to you, just validation that you do not appear to be doing anything wrong.  With a little luck, the post after this one will be someone swooping in to dunk on me and say, "No, you just need to do this...". 

Using the snip above, I can get to every ***individual*** value that has a ✅.  Everything with a ❌ requires me to get the parent and parse it.

```
{
✅    "kismet.device.base.macaddr": "AA:BB:CC:0F:47:D5",  
✅    "dot11.device": {
✅        "dot11.device.last_bssid": "AA:BB:CC:0F:47:D5",
✅       "dot11.device.responded_ssid_map": [
            {
❌                "dot11.advertisedssid.wps_device_name": "Xfinity Wireless Gateway",
            }
        ],
✅      "dot11.device.last_beacon_timestamp": 1727982007,
✅      "dot11.device.last_probed_ssid_record": {
✅          "dot11.probedssid.last_time": 1727981980,
✅          "dot11.probedssid.ssid": "MYSSID",
        },
✅        "dot11.device.num_associated_clients": 51,
✅      "dot11.device.advertised_ssid_map": [
            {
❌              "dot11.advertisedssid.ssidlen": 7,
❌              "dot11.advertisedssid.maxrate": 144.400000,
            }
        ],
✅      "dot11.device.client_disconnects": 1,
    },
✅    "kismet.device.base.manuf": "Vantiva USA LLC",
}
````

yurybelevskiy, 2024-10-04:
Hello @itdojo, thank you for the detailed response. I expected that it is something Kismet wouldn't support, but I wanted to double-check with the community, perhaps, there was some undocumented way of getting these fields.
At the moment, I am following the approach you suggested which is retrieving the entire parent structure and parsing out the field individually, yet when you are dealing with many access points and low bandwidth internet connection, my goal was to reduce traffic as I could by minimizing amount of fields sent.

RFDeck, 2026-11-07:
This is still the case in master and 2026-10-R1. The reason:

- `get_tracker_element_path()` walks the path with `path_as_map()`, which only steps into a plain `tracker_map`. The
  SSID maps (`advertised_ssid_map`, `responded_ssid_map`, `probed_ssid_map`) are `tracker_element_hashkey_map`, so
  the walk stops there.
- `summarize_tracker_element()` then puts in the generic placeholder for a missing field, which serializes as 0.
  That's the same placeholder used when a field just doesn't exist on a device, so the response gives no hint that
  the path itself can't work.

`get_tracker_element_multi_path()` can already walk vectors and maps; it's used for filters. Using it for `fields`
paths that go through a collection, and returning one value per entry, would cover this request. Would you take a
patch for that? If so, what shape should the result have: a list in place of the value, or a map keyed like the
original?

Until then, asking for the whole map (`dot11.device/dot11.device.responded_ssid_map`) and picking the field out on
the client side works.