kismetwireless/kismet #534: FATAL - Could not probe local source prior to connecting to the remote host

rhayy, 2024-09-14:
Hello the Kismet community,

### Intro
I'm facing an issue I can't resolve by myself.

I'm trying to set up a remote capture infrastucture between two hosts located on the same local network, both of them running kismet-2023-07-R1.

### Remote server
The first one is a Debian server with the following `kismet_site.conf` (everything else is defaults):
```
httpd_port=80
remote_capture_enabled=true
remote_capture_listen=0.0.0.0
remote_capture_port=8080
```
The server is launched with the following command line and output: 
```
# kismet
INFO: Including sub-config file: /etc/kismet/kismet_httpd.conf
INFO: Including sub-config file: /etc/kismet/kismet_memory.conf
INFO: Including sub-config file: /etc/kismet/kismet_alerts.conf
INFO: Including sub-config file: /etc/kismet/kismet_80211.conf
INFO: Including sub-config file: /etc/kismet/kismet_logging.conf
INFO: Including sub-config file: /etc/kismet/kismet_filter.conf
INFO: Including sub-config file: /etc/kismet/kismet_uav.conf
INFO: Loading config override file '/etc/kismet/kismet_package.conf'
INFO: Optional sub-config file not present: /etc/kismet/kismet_package.conf
INFO: Loading config override file '/etc/kismet/kismet_site.conf'
INFO: Loading optional sub-config file: /etc/kismet/kismet_site.conf
INFO: Setting server UUID 8F3A0C48-7297-11EF-AD67-4B49534D4554
INFO: Starting Beast webserver on 0.0.0.0:80
INFO: Opened OUI file '/usr/share/kismet/kismet_manuf.txt.gz
INFO: Indexing manufacturer db
INFO: Completed indexing manufacturer db, 34092 lines and 682 indexes.
INFO: Saving devices to the Kismet database log every 30 seconds
INFO: Using default rates of 10/min, 1/sec for alert 'DEVICEFOUND'
INFO: Using default rates of 10/min, 1/sec for alert 'DEVICELOST'
INFO: Registering support for DLT_PPI packet header decoding
INFO: Registering support for DLT_RADIOTAP packet header decoding
INFO: Registering support for DLT_BTLE_RADIO packet header decoding
INFO: Using default rates of 10/min, 1/sec for alert 'BADFIXLENIE'
INFO: PHY80211 will only process AP signal levels from beacons
INFO: Allowing Kismet clients to view WEP keys
INFO: Keeping EAPOL packets in memory for easy download and WIDS 
      functionality; this can use more RAM.
INFO: Registered PHY handler 'IEEE802.11' as ID 0
INFO: Registered PHY handler 'RFSENSOR' as ID 1
INFO: Registered PHY handler 'Z-Wave' as ID 2
INFO: Registered PHY handler 'Bluetooth' as ID 3
INFO: Registered PHY handler 'UAV' as ID 4
INFO: Registered PHY handler 'NrfMousejack' as ID 5
INFO: Using default rates of 10/min, 1/sec for alert 'BLEEDINGTOOTH'
INFO: Registered PHY handler 'BTLE' as ID 6
INFO: Registered PHY handler 'METER' as ID 7
INFO: Indexing ADSB ICAO db
INFO: Completed indexing ADSB ICAO db, 325554 lines 6512 indexes
INFO: Registered PHY handler 'ADSB' as ID 8
INFO: Registered PHY handler '802.15.4' as ID 9
INFO: Registered PHY handler 'RADIATION' as ID 10
INFO: Serving static file content from /usr/share/kismet/httpd/
INFO: Enabling channel hopping by default on sources which support channel 
      control.
INFO: Setting default channel hop rate to 5/sec
INFO: Enabling channel list splitting on sources which share the same list 
      of channels
INFO: Enabling channel list shuffling to optimize overlaps
INFO: Sources will be re-opened if they encounter an error
INFO: Saving datasources to the Kismet database log evert 30 seconds
INFO: Launching remote capture server on 0.0.0.0 8080
INFO: No data sources defined; Kismet will not capture anything until a 
      source is added.
INFO: Opened kismetdb log file './/Kismet-20240914-21-51-39-1.kismet'
INFO: Saving packets to the Kismet database log.
INFO: GPS track will be logged to the Kismet logfile
ALERT: ROOTUSER Kismet is running as root; this is less secure.  If you 
       are running Kismet at boot via systemd, make sure to use `systemctl 
       edit kismet.service` to change the user.  For more information, see 
       the Kismet README for setting up Kismet with minimal privileges.
INFO: Starting Kismet web server...
INFO: HTTP server listening on 0.0.0.0:80
```

### Sensor
The second one is a Kali-based sensor with a bluetooth adapter.
The server is launched with the following command line and output: 
```
$ sudo kismet_cap_linux_bluetooth --connect kismet.local:8080 --apikey $KISMET_API --source source=hci0
[2024/09/14 23:55:53:8559] N: lws_create_context: LWS: 4.3.3-unknown, NET CLI SRV H1 H2 WS ConMon IPV6-on
[2024/09/14 23:55:53:8561] N: __lws_lc_tag:  ++ [wsi|0|pipe] (1)
[2024/09/14 23:55:53:8562] N: __lws_lc_tag:  ++ [vh|0|netlink] (1)
[2024/09/14 23:55:53:8563] N: __lws_lc_tag:  ++ [vh|1|default||-1] (2)
FATAL - Could not probe local source prior to connecting to the remote host: 
FATAL:  Datasource exiting libwebsocket loop
INFO: capture process exited 0 signal 0
INFO: Sleeping 5 seconds before attempting to reconnect to remote server
[2024/09/14 23:55:58:8577] N: lws_create_context: LWS: 4.3.3-unknown, NET CLI SRV H1 H2 WS ConMon IPV6-on
[2024/09/14 23:55:58:8578] N: __lws_lc_tag:  ++ [wsi|0|pipe] (1)
[2024/09/14 23:55:58:8578] N: __lws_lc_tag:  ++ [vh|0|netlink] (1)
[2024/09/14 23:55:58:8579] N: __lws_lc_tag:  ++ [vh|1|default||-1] (2)
FATAL - Could not probe local source prior to connecting to the remote host: 
FATAL:  Datasource exiting libwebsocket loop
INFO: capture process exited 0 signal 0
INFO: Sleeping 5 seconds before attempting to reconnect to remote server
```

### Issue
Once everything is launched, the sensor prints an error in its logs :
```
FATAL - Could not probe local source prior to connecting to the remote host: 
FATAL:  Datasource exiting libwebsocket loop
```

However, running kismet on this same host is successful as shown here:
```
$ sudo kismet -c hci0
INFO: Including sub-config file: /etc/kismet/kismet_httpd.conf
[...]
INFO: Probing interface 'hci0' to find datasource type
[...]
INFO: HTTP server listening on 0.0.0.0:2501
INFO: Detected new Bluetooth device 5C:EA:1D:CB:C7:66
INFO: Detected new Bluetooth device 44:5C:E9:CB:77:05
INFO: Detected new Bluetooth device F1:83:F6:0A:6B:C4
INFO: Detected new Bluetooth device F4:93:1C:BD:7E:B9 (Daikin)
INFO: Detected new Bluetooth device 51:94:BF:4B:CD:67
INFO: Detected new Bluetooth device 63:6E:BE:92:40:4C
INFO: Detected new Bluetooth device 63:45:C4:AD:25:80
INFO: Detected new Bluetooth device D8:9C:67:74:7B:A4
INFO: Detected new Bluetooth device 59:19:57:3A:F3:DD
INFO: Detected new Bluetooth device 73:B3:5D:0C:C0:B4
```

Correct me if I am wrong: are the logs saying that it can't find any the `hci0` device ?

Stratos-Analytics, 2025-01-05:
sudo kismet_cap_linux_bluetooth --connect 127.0.0.1:2501 --apikey <>  --source hci0:name=1_Bluetooth

Works when i use the above command method

ondrejmo, 2025-01-19:
> sudo kismet_cap_linux_bluetooth --connect 127.0.0.1:2501 --apikey <> --source hci0:name=1_Bluetooth
> 
> Works when i use the above command method

Yeah, you need to remove the `source=` prefix (e.g. `--source source=hci0:name=harald` -> `--source hci0:name=harald`).