kismetwireless/kismet #548: Historical view of channel activity does not work in browser

martin3000, 2024-12-30:
Kismet 2023-07-R1 for Ubuntu 22.04, Installed from deb package.

If I select "Channels", "Historical","Past Minute", the bar stays on the left and the values are not propagated over the time.

[image]

Also, the API endpoint /channels/channels.json returns only zeros for the channel activity.

RFDeck, 2026-11-04:
The channel tracker and its history were rewritten after 2023-07-R1, and 5f190b9a ("channels: count packets, data and devices per frequency correctly") fixed how packets, data and devices are counted per frequency. Both are in 2026-10-R1.

Checked on current master, which has the same channel code as 2026-10-R1, with a pcapfile source replaying Wi-Fi in realtime: after 30 s, /channels/channels.json has non-zero values in the minute history for packets and devices, and the history moves forward with the current time.

Could you check whether the historical view works for you with 2026-10-R1 or later?

kismetwireless, 2026-11-04:
> The channel tracker and its history were rewritten after 2023-07-R1, and [5f190b9](https://github.com/kismetwireless/kismet/commit/5f190b9a67e9b2241e8e22298df3bac56df44bb4) ("channels: count packets, data and devices per frequency correctly") fixed how packets, data and devices are counted per frequency. Both are in 2026-10-R1.

Actually looks like this was broken in 2025-09 and didn't get fixed in 2026-10. Should be fine on master, already, in the new UI.

-m