kismetwireless/kismet #550: Build Failure: GCC15

limburgher, 2025-02-06:
On Fedora 42+:

In file included from libwifiuserspace/kernel/nl80211.h:36,
                 from libwifiuserspace/userspace/userspace.h:19,
                 from capture_hak5_coconut.c:55:
libwifiuserspace/kernel/types.h:26:18: error: conflicting types for ‘__u64’; have ‘uint64_t’ {aka ‘long unsigned int’}
   26 | typedef uint64_t __u64;
      |                  ^~~~~
In file included from /usr/include/asm-generic/types.h:7,
                 from /usr/include/asm/types.h:1,
                 from /usr/include/linux/types.h:5,
                 from /usr/include/linux/sched/types.h:5,
                 from /usr/include/bits/sched.h:63,
                 from /usr/include/sched.h:43,
                 from /usr/include/pthread.h:22,
                 from capture_hak5_coconut.c:35:
/usr/include/asm-generic/int-ll64.h:31:42: note: previous declaration of ‘__u64’ with type ‘__u64’ {aka ‘long long unsigned int’}
   31 | __extension__ typedef unsigned long long __u64;
      |                                          ^~~~~
libwifiuserspace/kernel/types.h:40:15: error: conflicting types for ‘__le64’; have ‘__u64’ {aka ‘long unsigned int’}
   40 | typedef __u64 __le64;
      |               ^~~~~~
/usr/include/linux/types.h:35:25: note: previous declaration of ‘__le64’ with type ‘__le64’ {aka ‘long long unsigned int’}
   35 | typedef __u64 __bitwise __le64;
      |                         ^~~~~~
libwifiuserspace/kernel/types.h:44:15: error: conflicting types for ‘__be64’; have ‘__u64’ {aka ‘long unsigned int’}
   44 | typedef __u64 __be64;
      |               ^~~~~~
/usr/include/linux/types.h:36:25: note: previous declaration of ‘__be64’ with type ‘__be64’ {aka ‘long long unsigned int’}
   36 | typedef __u64 __bitwise __be64;
      |                         ^~~~~~
In file included from libwifiuserspace/kernel/ieee80211.h:29,
                 from libwifiuserspace/kernel/cfg80211.h:22,
                 from wifi_coconut/wifi_coconut.h:27,
                 from capture_hak5_coconut.c:56:

RFDeck, 2026-11-05:
This should be fixed by 82e0ca25 ("coconut: rename all types from kernel types to stdint or otherwise custom types"). libwifiuserspace/kernel/types.h no longer defines __u64, __le64 or __be64, so it can't clash with <linux/types.h>, which now comes in through pthread.h. The commit is in 2025-09-R1 and 2026-10-R1, but not in 2023-07-R2, which may be what the Fedora build was based on.

Since you offered to test: does 2026-10-R1 build for you with --enable-wifi-coconut on Fedora 42?

limburgher, 2026-11-05:
2025-09-R1 built on Fedora 42 and onward, however 2026-10-R1 only builds on i686 and x86_64 for Fedora 46, not aarch64, ppc64le, and s390x as it typically does.  Would you like a separate issue for that?

RFDeck, 2026-11-05:
Thanks for testing. So the original type clash is fixed, and this issue can be closed.

A separate issue for 2026-10-R1 on aarch64, ppc64le and s390x would be good, with the first error from the build log
and the directory it's in. All three architectures have an unsigned `char`, so I built the server sources of
2026-10-R1 on x86_64 with `-funsigned-char`: all 175 C++ files compile with no errors or warnings. I'd guess the
problem is in one of the capture helpers added since 2025-09-R1 (cell, userspace USB Wi-Fi, HaLow, sniffle, and
others), but without the log that's only a guess.

limburgher, 2026-11-06:
Excellent.  https://github.com/kismetwireless/kismet/issues/717

closed 2026-11-06 by limburgher