kismetwireless/kismet #551: Crash (after running for 20 min or so) on Raspberry Pi 5

NHMatt2, 2025-02-07:
$ Uncaught exception "basic_string::substr: __pos (which is 3) > this->size() (which is 0)"
Stack trace (most recent call last) in thread 21767:
#8    Object "[0xffffffffffffffff], at 0xffffffffffffffff, in
#7    Object "/lib/aarch64-linux-gnu/libc.so.6, at 0x7faedaed8b, in
#6    Object "/lib/aarch64-linux-gnu/libc.so.6, at 0x7faed45453, in
#5    Object "/lib/aarch64-linux-gnu/libstdc++.so.6, at 0x7faf9ca37f, in
#4    Object "kismet, at 0x5568a5603b, in
#3    Object "/lib/aarch64-linux-gnu/libstdc++.so.6, at 0x7faf9994e7, in std::rethrow_exception(std::__exception_ptr::exception_ptr)
#2    Object "/lib/aarch64-linux-gnu/libstdc++.so.6, at 0x7faf990a63, in std::terminate()
#1    Object "/lib/aarch64-linux-gnu/libstdc++.so.6, at 0x7faf99950f, in
#0    Object "kismet, at 0x5568a5437b, in TerminationHandler()
Uncaught exception "std::exception"
FATAL:  Error during write(): Broken pipe

NHMatt2, 2025-02-07:
2023.07.R1 release on up-to-date kali (kali-rolling) machine (Linux kali-raspberrypi 6.6.63-v8+ #1 SMP PREEMPT Tue Dec  3 18:51:49 UTC 2024 aarch64 GNU/Linux)

kismetwireless, 2025-02-07:
I'd suggest trying nightlies; there have been significant changes since the last release, which are queued up for the next big release.

-m

NHMatt2, 2025-02-08:
Current installation was via package.  How would you recommend installing the nightly code from github?  Should I remove the installed package first, then build from source?  Or... should I compile the source off to the side and just drop in one or more files from the built version?  It has been ages since I've tried to incorporate "both" for a running environment.

kismetwireless, 2025-02-08:
There’s nightly packages on the kismet site for kali as a repo

NHMatt2, 2025-02-08:
Oh.  I see it there now,  Thanks much!