kismetwireless/kismet #569: Uncaught exception "basic_string::substr: __pos (which is 3) > this->size() (which is 2)

jimg4kfz, 2025-09-07:
H All, Im running Kali and Kismet on a fairly old Toshiba laptop with all the latest updates I can find, nightly’s etc.
Im using a TOPGNSS 803G GPS receiver and an ALFA AWUS1900 AP
The problem Im getting is outlined below after an error message appears on the Kali server, which runs OK for approx 10 minutes then stops with below error message.

(GPS) ERROR reading NMEA data: End of file. —  and on Kismet UI.

Im not an expert in any way with Linux, just trying to keep the grey matter ticking. (Im 78).

Ive looked on Github Kismetwireless and it seems this has been outlined before as an error. I note other issues have also reported this:
https://github.com/kismetwireless/kismet/issues/551

Not sure if this is a Kali or a Kismet problem, although it does feel like the interaction between being GPS data via whatever means (serial / USB / pipe / sockt ? ) might be the root cause i.e. “End of file” which I guess MIGHT be either CTRL-D delivered or the connection went away for some other reason.

BTW I have an RPi5 install that runs 100% without error.

Would appreciate any help taking into account Im no expert. 

Uncaught exception "basic_string::substr: __pos (which is 3) > this->size() (which is 2)"
Stack trace (most recent call last) in thread 3799:
#8    Object "[0xffffffffffffffff], at 0xffffffffffffffff, in 
#7    Object "/lib/x86_64-linux-gnu/libc.so.6, at 0x7ff638f1a7b7, in 
#6    Object "/lib/x86_64-linux-gnu/libc.so.6, at 0x7ff638e9cb7a, in 
#5    Object "/lib/x86_64-linux-gnu/libstdc++.so.6, at 0x7ff639ce6d63, in 
#4    Object "kismet, at 0x55ef6aac1850, in 
#3    Object "/lib/x86_64-linux-gnu/libstdc++.so.6, at 0x7ff639cb914c, in std::rethrow_exception(std::__exception_ptr::exception_ptr)
#2    Object "/lib/x86_64-linux-gnu/libstdc++.so.6, at 0x7ff639ca662d, in std::terminate()
#1    Object "/lib/x86_64-linux-gnu/libstdc++.so.6, at 0x7ff639cb9159, in 
#0    Object "kismet, at 0x55ef6aabf82a, in TerminationHandler()
Uncaught exception "std::exception"
FATAL:Error during write(): Broken pipe
zsh: IOT instruction  sudo kismet

dragorn, 2025-09-07:
> jimg4kfz created an issue [(kismetwireless/kismet#569)](https://github.com/kismetwireless/kismet/issues/569)

threw a possible fix into the latest git; it should go into tonights nightly build. not sure if it will address it or not.

RFDeck, 2026-11-05:
The old NMEA parser (`gpsnmea_v2.cc`) called `substr(3)` on the first field of a sentence without checking its length,
so a cut-off sentence with only two characters before the first comma threw exactly this exception. fc27d809
(2025-09-07) rejects such sentences, and since 23d52606 (2026-10-09) Kismet uses a new NMEA parser (`gpsnmea_v3.cc`)
that checks the length before every substring. Both are in 2026-10-R1.

Could you check whether the crash is gone with 2026-10-R1 or later?